analogdevicesinc/Pixelpulse2 #36: Build instructions dont make sense at all on a Mac

ralphrmartin, 2015-02-13:
(1) Setting up  Qt to build stuff on a mac needs a download of 700Mb for the Qt installer, and much more space to install Qt itself.

(2) The following instructions 
    git clone --recursive https://github.com/signalspec/pixelpulse2
    mkdir build
    qmake ..
say to get the sources, to make a build directly (surely that should be inside the pixlespulse directory, not at the same level as it) and then to call qmake on .., which just gives Error processing project file .. (surely that is the wrong directory).

Can't you just provide a binary? I want to do some hardware hacking, not to debug someone else's build instructions, and install tons of of irrelevant Qt stuff.

I thought I was buying a commercial project with some software in the ADALM1000. I'm very disappointed.

itdaniher, 2015-02-13:
Ralph - sorry to hear of your issues - I've updated the README to fix the instructions. I left ambiguity about the correct path. `qmake ..` should be run inside `pixelpulse2/build`.

I'll see what I can do about a binary for OSX.

rgetz, 2015-02-14:
Yeah, someone shipped units before the software went through its full QA. They were not supposed to. 

Sorry for the inconvience. As Ian said - we are working on it, and will get binaries posted next week.

hatchetation, 2015-02-14:
As someone who ended up with one of those pre-shipped units, it's hard to complain too much. :) 

FWIW, I'm having trouble building as well, with libusb and qt5 installed using homebrew:

    Project ERROR: libusb-1.0 development package not found

I monkeyed around with `QMAKE_` flags for a while, and got further -- but the build died with a gnarlier error: 

```
Undefined symbols for architecture x86_64:
  "_libusb_alloc_transfer", referenced from:
      Transfers::alloc(unsigned int, libusb_device_handle*, unsigned char, unsigned char, unsigned long, unsigned int, void (*)(libusb_transfer*), void*) in device_m1000.o
```

It would be nice to have happy-path build instructions (using homebrew or similar), even if a binary is coming soon.

ralphrmartin, 2015-02-14:
OK, now I understand what has happened. I'll try to be patient; thanks for updating the build instructions. But it would be good if you could provide a binary for MacOSX, or failing that, at least make a homebrew formula so it is easy to install, as the previous poster implies.

shabaz123, 2015-02-15:
Hi!
I wrote some instructions (kind-of "getting started guide") here, I hope this helps:

http://www.element14.com/community/groups/test-and-measurement/blog/2015/02/14/getting-started-with-the-active-learning-module-adalm1000

ragingSloth, 2015-02-15:
@hatchetation I think you're missing the development headers. Did you `brew install libusb-devel` or something similar or only `libusb`?

ralphrmartin, 2015-02-16:
(1) I cannot install on a Mac, as Qt5 is iincompatible with Qt4, and I need Qt4 for other software.

(2) Instructions still do not work on Linux . If you follow the ones here (not Shabaz's,as I have Debian), you just get 
bash: qmake: command not found
even after installing qt5 using the installer. Fixing that up by adding
/opt/Qt5.4.0/5.4/gcc_64/bin to the PATH
still doesn't work, as I then get told
libusb-1.0 development package not found
even though it is installed. This sounds similar to the error encountered when using homebrew on the Mac.

itdaniher, 2015-02-24:
Hi All,

We have a preliminary build for OSX at https://github.com/analogdevicesinc/pixelpulse2/releases/tag/macosx. 

Work documenting the build process and (hopefully) automating the creation of packages is ongoing, but feel free to try out the existing .dmg and .app.

Thanks,
Ian

ralphrmartin, 2015-02-24:
> On 24 Feb 2015, at 16:56, Ian Daniher <notifications@github.com> wrote:
> We have a preliminary build for OSX at https://github.com/analogdevicesinc/pixelpulse2/releases/tag/macosx.
> 
> Work documenting the build process and (hopefully) automating the creation of packages is ongoing, but feel free to try out the existing .dmg and .app.

Thanks. Will try it, and let you know if there are any problems.

Best wishes

Ralph
--
Prof. Ralph Martin                     Phone: +44(0)29 2087 5536
Computer Science & Informatics         Fax:   +44(0)29 2087 4598
Cardiff University                 Email: martinrr@cardiff.ac.uk
5 The Parade, Roath              WWW:  http://ralph.cs.cf.ac.uk/
Cardiff, CF24 3AA, UK         VOIP: sip:ralphmartin@sip2sip.info

itdaniher, 2015-03-03:
Anyone get a chance to play with the Pixelpulse2 OSX dmg?

ralphrmartin, 2015-03-03:
> On 3 Mar 2015, at 15:14, Ian Daniher <notifications@github.com> wrote:
> 
> Anyone get a chance to play with the Pixelpulse2 OSX dmg?

Yes. Crashes at launch as it cannot find Qt5.

I am unable to install Qt5 as it conflicts with Qt4, and other software I need depends on Qt4.

Best wishes

Ralph
--
Prof. Ralph Martin                     Phone: +44(0)29 2087 5536
Computer Science & Informatics         Fax:   +44(0)29 2087 4598
Cardiff University                 Email: martinrr@cardiff.ac.uk
5 The Parade, Roath              WWW:  http://ralph.cs.cf.ac.uk/
Cardiff, CF24 3AA, UK         VOIP: sip:ralphmartin@sip2sip.info

itdaniher, 2015-03-03:
Thanks for the feedback. @radhermit and I will see what we need to do to package Qt5 depens with the OSX build.

radhermit, 2015-03-03:
Try the bundled version (https://github.com/analogdevicesinc/pixelpulse2/releases/download/macosx/pixelpulse2-bundled.dmg.zip) that I tossed together now. It should include all required the qt5/libusb within the application sandbox.

ralphrmartin, 2015-03-04:
> On 3 Mar 2015, at 22:10, Ian Daniher <notifications@github.com> wrote:
> 
> Try the bundled version (https://github.com/analogdevicesinc/pixelpulse2/releases/download/macosx/pixelpulse2-bundled.dmg.zip) that I tossed together now. It should include all required the qt5/libusb within the application sandbox.

Hooray. It works.

It's almost impossible to read the axes on the plots though - dark gray on black is not the best combination. Can we please have white on black for ALL the annotations?

ralphrmartin, 2015-03-04:
Hmm, spoke too soon.

After switching from sine wave on voltage to sawtooth or square wave, the blue buttons no longer allow me to drag the amplitude of the wave up and down, and indeed, a third blue button appears. Something isn't working quite right.

itdaniher, 2015-03-04:
@ralphrmartin Thanks for the suggestion, I noted your preference for white on black at https://github.com/analogdevicesinc/pixelpulse2/issues/5. I'll make some adjustments to improve readability before our initial public release.

Closing this issue as hopefully we have have spared you and all future OSX users the need to compile from source.

closed 2015-03-04 by itdaniher

ralphrmartin, 2015-03-04:
Yes, thanks for providing a binary. Just needs debugging...

hatchetation, 2015-03-04:
@radhermit thanks for the OS X binary - works for me here, seems to be functionally equiv. to a local linux build. 

Sorry for the radio silence (busy!), looking forward to diving in.