analogdevicesinc/Pixelpulse2 #42: Phone home doesnt exist

dNechita, 2015-02-19:
Need a way to see if new applications/library/dirvers should be installed. This can be done easily with curl.

Is there a settings or ini or other type of persistent storage in pixelpulse to save the setting "Don't bug me again"?

Thanks

itdaniher, 2015-02-19:
Pixelpulse2's build system defines GIT_VERSION and BUILD_DATE at compile time. A naive implementation could check the GIT_VERSION shorttag and compare against GH's latest tag.

rgetz, 2015-02-21:
Assuming that we are following something like: https://help.github.com/articles/creating-releases/ and using  semantic versioning, (which so far we have not done -- correct?)

I can see two things:
 - check a specific wiki page for install versions. For example:
http://wiki.analog.com/university/tools/m1k/pixelpulse/latest_release
- check github for a new release tag on the main branch (and assume that we have builds for all the architectures when this is done, which may or maynot be true).
https://developer.github.com/v3/repos/releases/#get-the-latest-release

Either way - I'm not sure if a git commit hash is the best way to manage releases. A MAJOR.MINOR.PATCH version number is the way to go (IMHO). For pre-release/testing - MAJOR.MINOR.PATCH-rc.hash seems like it would work the best.

Thoughts?

-Robin

dNechita, 2015-02-26:
I checked out the links that Robin posted and I implemented the "phone-home" functionality mainly by requesting the list of releases of the github repository: pixelpulse2. The build date of the latest release is being compared with the build date of client(the build date is not yet stored anywhere).

The dependencies are libcurl and libjansson. How can I add them so that when I commit my changes the application will build without dependency errors?

Thanks,
-Dan

rgetz, 2015-02-26:
I don't think it is a problem -- but we need to make sure both are available with mingw.

kevinmehall, 2015-02-26:
You can use the HTTP and JSON functionality built into Qt instead of adding dependencies:
http://doc.qt.io/qt-5/topics-network-connectivity.html
http://doc.qt.io/qt-5/json.html

rgetz, 2015-02-26:
Dan:

Yeah, Lars and I were discussing over dinner. -- Kevin is correct - it would be better to use the built in libs/functions. No point in adding extra dependencies into Qt that don't need to be there. 

I think you can take the code you have, and use it in osc with:
https://developer.gnome.org/json-glib/stable/
And the existing curl code. 

- Robin

closed 2015-02-26 by rgetz

dNechita, 2015-03-06:
The "phone-home" branch is now on git. For now it will use libcurl and libjansson dependencies.

dNechita, 2015-03-12:
I've added a Qt implementation of "phone-home" on the qt_phonehome branch. This can replace the libcurl implementation.