analogdevicesinc/Pixelpulse2 #44: build issues on debian stable (wheezy)

dNechita, 2015-02-19:
analog@CJLAB-D01:~/git/pixelpulse2/build$ cat /etc/issue
Debian GNU/Linux 7 \n \l
analog@CJLAB-D01:~/git/pixelpulse2/build$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 7.6 (wheezy)
Release:	7.6
Codename:	wheezy

analog@CJLAB-D01:~/git/pixelpulse2/build$ sudo apt-get install libusb-1.0-0-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libusb-1.0-0-dev is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 130 not upgraded.
analog@CJLAB-D01:~/git/pixelpulse2/build$ qmake ..
sh: 1: pkg-config: not found
Project ERROR: Package libusb-1.0 not found

itdaniher, 2015-02-19:
`apt-get install pkg-config`

itdaniher, 2015-02-19:
To verify functionality, try `pkg-config --list-all | grep usb`

itdaniher, 2015-02-19:
To collect the needed dependencies, try: 

> sudo apt-get install qtdeclarative5-dev mesa-common-dev libusb-1.0-dev build-essential pkg-config qt5-default qtbase5-dev

itdaniher, 2015-02-19:
Once this is resolved, could you please verify the names of the dependencies required on Debian 7?

dNechita, 2015-02-20:
Packages needed from the backports repository (deb http://ftp.debian.org/debian wheezy-backports main contrib non-free):
qtdeclarative5-dev
qt5-default qtbase5-dev

Now I get the following errors:
	analog@CJLAB-D01:~/git/pixelpulse2/build$ make
	make -f Makefile.Release
	make[1]: Entering directory `/home/analog/git/pixelpulse2/build'
	g++ -c -m64 -pipe -O2 -std=c++0x -Wall -W -D_REENTRANT -fPIE -DGIT_VERSION="\"5547a9b\"" -DBUILD_DATE="\"Fri Feb 20 11:52:43 EET 2015\"" -DQT_NO_DEBUG -DQT_QUICK_LIB -DQT_QML_LIB -DQT_WIDGETS_LIB -DQT_NETWORK_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -I../../pixelpulse2 -isystem /usr/include/libusb-1.0 -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtQuick -isystem /usr/include/x86_64-linux-gnu/qt5/QtQml -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -Irelease -I. -o release/session.o ../libsmu/session.cpp
	../libsmu/session.cpp:18:46: error: ‘libusb_hotplug_event’ has not been declared
	../libsmu/session.cpp: In constructor ‘Session::Session()’:
	../libsmu/session.cpp:29:28: error: ‘LIBUSB_CAP_HAS_HOTPLUG’ was not declared in this scope
	../libsmu/session.cpp:32:5: error: ‘libusb_hotplug_event’ was not declared in this scope
	../libsmu/session.cpp:32:27: error: ‘LIBUSB_HOTPLUG_EVENT_DEVICE_ARRIVED’ was not declared in this scope
	../libsmu/session.cpp:32:65: error: ‘LIBUSB_HOTPLUG_EVENT_DEVICE_LEFT’ was not declared in this scope
	../libsmu/session.cpp:33:5: error: ‘libusb_hotplug_flag’ was not declared in this scope
	../libsmu/session.cpp:34:4: error: ‘LIBUSB_HOTPLUG_MATCH_ANY’ was not declared in this scope
	../libsmu/session.cpp:40:3: error: ‘libusb_hotplug_register_callback’ was not declared in this scope
	../libsmu/session.cpp: At global scope:
	../libsmu/session.cpp:93:49: error: ‘libusb_hotplug_event’ has not been declared
	../libsmu/session.cpp: In function ‘int hotplug_callback_usbthread(libusb_context*, libusb_device*, int, void*)’:
	../libsmu/session.cpp:96:15: error: ‘LIBUSB_HOTPLUG_EVENT_DEVICE_ARRIVED’ was not declared in this scope
	../libsmu/session.cpp:98:22: error: ‘LIBUSB_HOTPLUG_EVENT_DEVICE_LEFT’ was not declared in this scope
	../libsmu/session.cpp: In member function ‘void Session::progress()’:
	../libsmu/session.cpp:262:26: error: ‘UINT64_MAX’ was not declared in this scope
	make[1]: *** [release/session.o] Error 1
	make[1]: Leaving directory `/home/analog/git/pixelpulse2/build'
	make: *** [release] Error 2

dNechita, 2015-02-20:
FYI: Installing libusb from: https://github.com/kevinmehall/libusb/tree/hp fixed the libusb errors except ../libsmu/session.cpp:262:26: error: ‘UINT64_MAX’ was not declared in this scope

dNechita, 2015-02-20:
Starting fresh with:
analog@cjlab-d01:~$ cat /etc/issue
Ubuntu 14.04.2 LTS \n \l

Build went fine. I should mention

But could not run the app:
analog@cjlab-d01:~/git/analogdevicesinc/pixelpulse2/build$ ./pixelpulse2 
Using libusb hotplug
QQmlApplicationEngine failed to load component 
qrc:/qml/main.qml:4 module "QtQuick.Controls" is not installed
qrc:/qml/main.qml:3 module "QtQuick.Layouts" is not installed
qrc:/qml/main.qml:4 module "QtQuick.Controls" is not installed
qrc:/qml/main.qml:3 module "QtQuick.Layouts" is not installed

dNechita, 2015-02-20:
I defined UINT64_MAX in order to continue the compilation and I noticed that there is one more dependency that needs to be installed:
qml-module-qtquick-controls

itdaniher, 2015-02-23:
There's a new branch 'debian' which implements a first draft at packaging pixelpulse2 for Debian Testing / Ubuntu 14.04+. It's not quite finished, but will be very easy to make available via PPA.

pcercuei, 2015-02-23:
I pushed a "debian_jessie" branch with some fixes to get it to compile on my PC.

itdaniher, 2015-02-23:
I have verified that the debian_jessie patches don't break anything on Ubuntu 12.04+. Merging changes into the debian branch, thanks @pcercuei !

itdaniher, 2015-02-26:
@rgetz et al, can you verify the debian package builds on stable? Then we should add manpages / links, merge the debian packaging metadata, close this issue, and set up an Ubuntu PPA.

itdaniher, 2015-03-03:
Confirming that PP2 builds on Debian Stable. Backports is required as Qt5 seems not to be offered for the current Stable release. 

I modified LibSMU at https://github.com/signalspec/libsmu/commit/febbea1f98decb11d6164172015084cd5d740828 to solve the build error mentioned above - UINT64_MAX is a C++11 #define that has an easy backwards-compatible replacement.

The fact that LibUSB-1.0-0-dev in Stable does not support USB Hotplug events does require a compatible LibUSB to be compiled from source. This means that the build protocol for all modern distros will be different than the build protocol for Debian Stable, but no aspect of the current codebase precludes its compilation on GCC-4.7 etc.

To build on Debian Stable, build/install [LibUSB from source](http://sourceforge.net/projects/libusb/files/latest/download?source=files), run ```dh_make``` as per the README, and append the ```-d``` flag to ```dpkg_buildpackage``` to create a working .deb package without the broken libusb-1.0-0.dev package installed on your system.

With Debian Stable compat confirmed, adding docs and closing PR #57.

closed 2015-03-03 by itdaniher