analogdevicesinc/Pixelpulse2 #46: Runtime error initializing device

dNechita, 2015-02-20:
On Linux, when I run the application with the adalm1000 connected:

(gdb) r
Starting program: /home/analog/git/analogdevicesinc/pixelpulse2/build/pixelpulse2 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffeb5cd700 (LWP 9212)]
[New Thread 0x7fffe0b8b700 (LWP 9213)]
[New Thread 0x7fffdbfff700 (LWP 9214)]
[New Thread 0x7fffdb7fe700 (LWP 9216)]
[New Thread 0x7fffda7fd700 (LWP 9217)]
Using libusb hotplug
[New Thread 0x7fffd9ffc700 (LWP 9218)]
Error initializing device

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7bc7ee1 in libusb_close () from /lib/x86_64-linux-gnu/libusb-1.0.so.0
(gdb) backtrace full
#0  0x00007ffff7bc7ee1 in libusb_close () from /lib/x86_64-linux-gnu/libusb-1.0.so.0
No symbol table info available.
#1  0x000000000043d182 in Device::~Device (this=0x82e8d0, __in_chrg=<optimized out>) at ../libsmu/session.cpp:289
No locals.
#2  0x0000000000438ed7 in M1000_Device::~M1000_Device (this=0x82e8d0, __in_chrg=<optimized out>) at ../libsmu/device_m1000.cpp:65
No locals.
#3  0x0000000000438f0c in M1000_Device::~M1000_Device (this=0x82e8d0, __in_chrg=<optimized out>) at ../libsmu/device_m1000.cpp:65
No locals.
#4  0x0000000000440d66 in std::_Sp_counted_ptr<M1000_Device*, (__gnu_cxx::_Lock_policy)2>::_M_dispose (this=0x82ec90) at /usr/include/c++/4.8/bits/shared_ptr_base.h:290
No locals.
#5  0x0000000000430ab8 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x82ec90) at /usr/include/c++/4.8/bits/shared_ptr_base.h:144
No locals.
#6  0x000000000042df49 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=0x7fffffffdbe8, __in_chrg=<optimized out>) at /usr/include/c++/4.8/bits/shared_ptr_base.h:546
No locals.
#7  0x000000000042ddec in std::__shared_ptr<Device, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=0x7fffffffdbe0, __in_chrg=<optimized out>) at /usr/include/c++/4.8/bits/shared_ptr_base.h:781
No locals.
#8  0x000000000042de2c in std::shared_ptr<Device>::~shared_ptr (this=0x7fffffffdbe0, __in_chrg=<optimized out>) at /usr/include/c++/4.8/bits/shared_ptr.h:93
No locals.
#9  0x000000000043c868 in Session::probe_device (this=0x82db80, device=0x820a70) at ../libsmu/session.cpp:157
        dev = {<std::__shared_ptr<Device, (__gnu_cxx::_Lock_policy)2>> = {_M_ptr = 0x82e8d0, _M_refcount = {_M_pi = 0x82ec90}}, <No data fields>}
        desc = {bLength = 18 '\022', bDescriptorType = 1 '\001', bcdUSB = 512, bDeviceClass = 0 '\000', bDeviceSubClass = 0 '\000', bDeviceProtocol = 0 '\000', bMaxPacketSize0 = 64 '@', idVendor = 1611, 
          idProduct = 30796, bcdDevice = 256, iManufacturer = 1 '\001', iProduct = 2 '\002', iSerialNumber = 3 '\003', bNumConfigurations = 1 '\001'}
        r = 0
#10 0x000000000043c524 in Session::update_available_devices (this=0x82db80) at ../libsmu/session.cpp:120
        dev = {<std::__shared_ptr<Device, (__gnu_cxx::_Lock_policy)2>> = {_M_ptr = 0x0, _M_refcount = {_M_pi = 0x0}}, <No data fields>}
        i = 1
        list = 0x8243c0
        num = 10
#11 0x0000000000429aa3 in SessionItem::openAllDevices (this=0x7fffffffdda0) at ../SMU.cpp:54
No locals.
#12 0x000000000042900c in main (argc=1, argv=0x7fffffffdec8) at ../main.cpp:99
        app = <incomplete type>
        engine = <incomplete type>
        smu_session = {<QObject> = {<No data fields>}, static staticMetaObject = {d = {superdata = 0x7ffff69dbf80 <QObject::staticMetaObject>, stringdata = 0x4504c0 <qt_meta_stringdata_SessionItem>, 
              data = 0x4508a0 <qt_meta_data_SessionItem>, static_metacall = 0x442fe6 <SessionItem::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)>, relatedMetaObjects = 0x0, 
              extradata = 0x0}}, m_session = 0x82db80, m_active = false, m_continuous = false, m_sample_rate = 0, m_sample_count = 0, m_devices = {{p = {static shared_null = {ref = {atomic = {
                      _q_value = -1}}, alloc = 0, begin = 0, end = 0, array = {0x0}}, d = 0x7ffff6641cd0 <QListData::shared_null>}, d = 0x7ffff6641cd0 <QListData::shared_null>}}}
        r = 32767
(gdb)

itdaniher, 2015-02-20:
I believe this is due to calling the deconstructors on uninitialized devices after permissions error out. Tim's patch at https://github.com/signalspec/libsmu/pull/19/files#diff-579bdf1c41cb77270e445580711356afR288 ought to address this - I'll test.

itdaniher, 2015-02-20:
The shotgun (crude but effective) approach to solving this issue on linux is to construct

`/etc/udev/rules.d/70-smus.rules`

with the contents

    SUBSYSTEM=="usb", ATTR{idVendor}=="59e3", MODE="0666"
    SUBSYSTEM=="usb", ATTR{idVendor}=="0456", MODE="0666"
    SUBSYSTEM=="usb", ATTR{idVendor}=="064b", MODE="0666"
    SUBSYSTEM=="usb", ATTR{idVendor}=="03eb", MODE="0666"

After creating this file, `sudo udevadm control --reload-rules` will cause a refresh. Unplug and replug the M1K, and you should be good to go!

itdaniher, 2015-02-20:
Confirming Tim's patch addresses the segfault observed, added more subtle udev rules to the udev branch. More nuanced instructions at https://github.com/signalspec/pixelpulse2/tree/udev. Open pull request to merge these changes into master at https://github.com/signalspec/pixelpulse2/pull/47.

closed 2015-02-20 by itdaniher