analogdevicesinc/Pixelpulse2 #48: backtracing not built in

dNechita, 2015-02-20:
Need to do something that prints out backtraces:

Code exists at:
http://panthema.net/2008/0901-stacktrace-demangled/

But it needs to display in a QT Window.
https://qt-project.org/doc/qt-4.7/unix-signals.html

I have the first implemented, and will make a branch.

dNechita, 2015-02-20:
I've pushed the "backtracking" branch with my changes.

In order to actually see the symbols instead of addresses it is needed to add the compile flag:
-rdynamic to the CFLAGS, CXXFLAGS, LFLAGS.
Since the make file is being generated automatically, Ian can you make sure that the -rdynamic flag is also included?
Thanks!

itdaniher, 2015-02-21:
It [looks like](http://www.gnu.org/savannah-checkouts/gnu/gnulib/manual/html_node/execinfo_002eh.html) GLIBC's execinfo.h which provides backtrace and backtrace_symbols isn't available on Windows. 

@rgetz is there someone we should ask about implementing the windows-specific functionality?

rgetz, 2015-02-23:
Yeah, Dan will been to add some mingw specifics.

Might make sense to break out into a seperate file, and just call init_signal_handers() from main. That way main will not look so fugly.

It appears that mingw needs to use Window's StackWalk64 function, which is documented:
https://msdn.microsoft.com/en-us/library/windows/desktop/ms680650%28v=vs.85%29.aspx

There is an example that he can use at:
http://spin.atomicobject.com/2013/01/13/exceptions-stack-traces-c/
or
https://code.google.com/p/backtrace-mingw/source/browse/trunk/backtrace.c

but I don't think that we can assume that end users have 'addr2line' installed on their system (like the first example). For now, just the addresses might be all that is necessary - and we can manually run things through addr2line...

Thoughts?

-Robin

dNechita, 2015-02-25:
I've added the backtracing code for windows. I've followed the links above and I went with <imagehlp> and <dbghlp> libraries. I can print out the addresses only.
Unfortunately when I compiled the code with g++ instead of gcc it doesn't work.
I "get the application was unable to start correctly 0xc00007b" at run-time.

I'm currently looking for a fix.

dNechita, 2015-02-25:
Forgot to mention that
LDFLAGS should be added: -limagehlp -ldbghelp

itdaniher, 2015-02-25:
The latest build of this branch failed with an error at https://ci.appveyor.com/project/itdaniher/pixelpulse2/build/0.1.47#L15824. Compile logs can be browsed at https://ci.appveyor.com/project/itdaniher/pixelpulse2/history.

Are imagehlp / dbghelp built-ins on Windows, or do they need to be installed from an external source?

Thanks! Ian

dNechita, 2015-02-25:
There are built-in.

-Dan

itdaniher, 2015-02-25:
Trying with those libs included - https://github.com/analogdevicesinc/pixelpulse2/blob/backtracking/pixelpulse2.pro#L62.

Next build is at https://ci.appveyor.com/project/itdaniher/pixelpulse2/build/0.1.49.

dNechita, 2015-02-25:
Thank you.
The build completed successfully.
I'm going to check if it actually works.

dNechita, 2015-02-26:
Thank you Ian for adding the "crash" branch which allowed me to get the entire call stack when app crashes.

...pixelpulse2_r3\dist>pixelpulse2.exe
Using libusb hotplug
eval code:1:17: Expected token `identifier'
eval code:1:17: Expected token `identifier'
Error: EXCEPTION_ACCESS_VIOLATION
FrameAddr: 0x0
pixelpulse2: FrameAddr: 0x40C594
FrameAddr: 0x28FE78
FrameAddr: 0x28FE78
FrameAddr: 0x31CE638
FrameAddr: 0x324D108
FrameAddr: 0x3245D10
Qt5Core: FrameAddr: 0x68B2FBC0 symbol: ZN10QArrayData11shared_nullE

rgetz, 2015-02-26:
So, if you use addr2line by hand - does it point you to the right place?

rgetz, 2015-02-26:
For 0x40C594?

dNechita, 2015-02-26:
Unfortunatelly no. I ran:
addr2line -e pixelpulse2.exe 0x40C594
And I get:
??:0

dNechita, 2015-02-27:
I've added -g option flag and it should make things better.

rgetz, 2015-02-27:
Only if the build system doesn't strip things.

dNechita, 2015-03-02:
Ian,
When you created the "Debug" branch in order to build the debug version of pixelpulse2, the application could not start anymore. I don't know why. Below you can see the error that shows when launching the app:
[image]

Once we figure how to successfully launch the program we will be able to translate the call stack addresses into file-name and line-number using addr2line utility.

-Dan

itdaniher, 2015-03-02:
Qt5Cored.dll is the debug version of the Qt5 core library. 

To find it on your system:

> dir c:\Qt5Cored.dll /s

itdaniher, 2015-03-02:
Dan - I've packaged the debug dlls required for the latest build of your branch. I'm uploading them to http://itdaniher.com/static/pixelpulse2-debug-package.tar.gz. Wait another 5min and you should be good to go.

dNechita, 2015-03-09:
Ian,

Lately the work has been done on the "debug" branch so the "bracktracking" branch is a little behind.
Do you want me to update the "backtraking" branch and request a pull or should we use the "debug" branch?

dNechita, 2015-03-09:
Actually I think it's best to use the "backtracking" branch.
The "debug" branch has a lot of useless commits in addition to the code snippet that allowed me to crash the application with "session.crash()" command.

itdaniher, 2015-03-09:
Dan - I think the Backtracking branch, once updated against 'debug, will suffice.

closed 2015-03-13 by itdaniher