analogdevicesinc/Pixelpulse2 #54: Density gradiated?

kuldeepdhaka, 2015-02-23:
How is "OpenGL accelerated density-gradiated rendering" being done?
as per the source:
 https://github.com/analogdevicesinc/pixelpulse2/blob/master/Plot/PhosphorRender.cpp#L13
https://github.com/analogdevicesinc/pixelpulse2/blob/master/Plot/PhosphorRender.cpp#L22
as seen in source, their is gradient height gradient is not dependent on sample values.

reference: http://www.tek.com/blog/sample-processing-digital-oscilloscope

kevinmehall, 2015-02-23:
All samples are displayed as translucent points, and their brightness is additive with `glBlendFunc(GL_ONE, GL_ONE)`. Thus pixels containing more sample points are brighter.

itdaniher, 2015-02-24:
@kuldeepdhaka closing this issue with @kevinmehall's answer. Feel free to open additional issues questions and comments. Thanks!

closed 2015-02-24 by itdaniher