analogdevicesinc/Pixelpulse2 #58: triage / resolve valgrind-reported memcheck errors

itdaniher, 2015-02-24:
valgrind --leak-check=full ./pixelpulse2 2>log

reports the following...

```valgrind
==21494== Memcheck, a memory error detector
==21494== Copyright (C) 2002-2013, and GNU GPL'd, by Julian Seward et al.
==21494== Using Valgrind-3.9.0 and LibVEX; rerun with -h for copyright info
==21494== Command: ./pixelpulse2
==21494== 
Using libusb hotplug
device insert 0x177d4400
==22228== Warning: invalid file descriptor 1024 in syscall close()
==22228== Warning: invalid file descriptor 1025 in syscall close()
==22228== Warning: invalid file descriptor 1026 in syscall close()
==22228== Warning: invalid file descriptor 1027 in syscall close()
==22228==    Use --log-fd=<number> to select an alternative log fd.
==22228== Warning: invalid file descriptor 1028 in syscall close()
==22228== Warning: invalid file descriptor 1029 in syscall close()
Session::detached ser: 2031315434393235313830333330323
m_devices.size() 0
Session::attached ser: 2031315434393235313830333330323
0x17ac44b0
device insert 0x17ac44b0
0x17ac44b0
Closing devices
m_devices.size() 0
==21494== 
==21494== HEAP SUMMARY:
==21494==     in use at exit: 509,201 bytes in 5,450 blocks
==21494==   total heap usage: 977,102 allocs, 971,652 frees, 227,332,316 bytes allocated
==21494== 
==21494== 3 bytes in 1 blocks are definitely lost in loss record 2 of 876
==21494==    at 0x4C2745D: malloc (in /usr/lib64/valgrind/vgpreload_memcheck-amd64-linux.so)
==21494==    by 0x84350B9: strdup (strdup.c:42)
==21494==    by 0x14AD0CEF: ???
==21494==    by 0x14AD99BC: ???
==21494==    by 0x14AD9B0E: ???
==21494==    by 0x14202955: ???
==21494==    by 0x60752E0: ??? (in /opt/Qt5.4.0/5.4/gcc_64/lib/libQt5Gui.so.5.4.0)
==21494==    by 0x6078304: QFontDatabase::findFont(int, QFontPrivate const*, QFontDef const&, bool, bool) (in /opt/Qt5.4.0/5.4/gcc_64/lib/libQt5Gui.so.5.4.0)
==21494==    by 0x6079667: QFontDatabase::load(QFontPrivate const*, int) (in /opt/Qt5.4.0/5.4/gcc_64/lib/libQt5Gui.so.5.4.0)
==21494==    by 0x6050BF4: QFontPrivate::engineForScript(int) const (in /opt/Qt5.4.0/5.4/gcc_64/lib/libQt5Gui.so.5.4.0)
==21494==    by 0x6082003: QTextEngine::fontEngine(QScriptItem const&, QFixed*, QFixed*, QFixed*) const (in /opt/Qt5.4.0/5.4/gcc_64/lib/libQt5Gui.so.5.4.0)
==21494==    by 0x60833E6: QTextEngine::shapeText(int) const (in /opt/Qt5.4.0/5.4/gcc_64/lib/libQt5Gui.so.5.4.0)
==21494== 
==21494== 8 bytes in 1 blocks are possibly lost in loss record 24 of 876
==21494==    at 0x4C2745D: malloc (in /usr/lib64/valgrind/vgpreload_memcheck-amd64-linux.so)
==21494==    by 0x4C2932B: realloc (in /usr/lib64/valgrind/vgpreload_memcheck-amd64-linux.so)
==21494==    by 0x8C1EF16: g_realloc (in /usr/lib64/libglib-2.0.so.0.3800.2)
==21494==    by 0x89AAD98: ??? (in /usr/lib64/libgobject-2.0.so.0.3800.2)
==21494==    by 0x89AF8F4: g_type_register_static (in /usr/lib64/libgobject-2.0.so.0.3800.2)
==21494==    by 0x89B40B1: g_type_plugin_get_type (in /usr/lib64/libgobject-2.0.so.0.3800.2)
==21494==    by 0x898A494: ??? (in /usr/lib64/libgobject-2.0.so.0.3800.2)
==21494==    by 0x400F2D9: call_init.part.0 (dl-init.c:82)
==21494==    by 0x400F3C2: _dl_init (dl-init.c:34)
==21494==    by 0x4001229: ??? (in /usr/lib64/ld-2.18.so)
==21494== 
==21494== 8 bytes in 1 blocks are possibly lost in loss record 25 of 876
==21494==    at 0x4C2745D: malloc (in /usr/lib64/valgrind/vgpreload_memcheck-amd64-linux.so)
==21494==    by 0x4C2932B: realloc (in /usr/lib64/valgrind/vgpreload_memcheck-amd64-linux.so)
==21494==    by 0x8C1EF16: g_realloc (in /usr/lib64/libglib-2.0.so.0.3800.2)
==21494==    by 0x89AAD98: ??? (in /usr/lib64/libgobject-2.0.so.0.3800.2)
==21494==    by 0x89AF8F4: g_type_register_static (in /usr/lib64/libgobject-2.0.so.0.3800.2)
==21494==    by 0x898C5D1: g_boxed_type_register_static (in /usr/lib64/libgobject-2.0.so.0.3800.2)
==21494==    by 0x898C7B5: g_value_array_get_type (in /usr/lib64/libgobject-2.0.so.0.3800.2)
==21494==    by 0x899DCD4: ??? (in /usr/lib64/libgobject-2.0.so.0.3800.2)
==21494==    by 0x898A4BE: ??? (in /usr/lib64/libgobject-2.0.so.0.3800.2)
==21494==    by 0x400F2D9: call_init.part.0 (dl-init.c:82)
==21494==    by 0x400F3C2: _dl_init (dl-init.c:34)
[... 1932 lines cut ...]
==21494==    at 0x4C2745D: malloc (in /usr/lib64/valgrind/vgpreload_memcheck-amd64-linux.so)
==21494==    by 0x14ACAC7A: ???
==21494==    by 0x14AD721C: ???
==21494==    by 0x14AD7295: ???
==21494==    by 0x14AD731A: ???
==21494==    by 0x14202848: ???
==21494==    by 0x60752E0: ??? (in /opt/Qt5.4.0/5.4/gcc_64/lib/libQt5Gui.so.5.4.0)
==21494==    by 0x6078304: QFontDatabase::findFont(int, QFontPrivate const*, QFontDef const&, bool, bool) (in /opt/Qt5.4.0/5.4/gcc_64/lib/libQt5Gui.so.5.4.0)
==21494==    by 0x6079667: QFontDatabase::load(QFontPrivate const*, int) (in /opt/Qt5.4.0/5.4/gcc_64/lib/libQt5Gui.so.5.4.0)
==21494==    by 0x6050BF4: QFontPrivate::engineForScript(int) const (in /opt/Qt5.4.0/5.4/gcc_64/lib/libQt5Gui.so.5.4.0)
==21494==    by 0x6082003: QTextEngine::fontEngine(QScriptItem const&, QFixed*, QFixed*, QFixed*) const (in /opt/Qt5.4.0/5.4/gcc_64/lib/libQt5Gui.so.5.4.0)
==21494==    by 0x60833E6: QTextEngine::shapeText(int) const (in /opt/Qt5.4.0/5.4/gcc_64/lib/libQt5Gui.so.5.4.0)
==21494== 
==21494== LEAK SUMMARY:
==21494==    definitely lost: 32,645 bytes in 602 blocks
==21494==    indirectly lost: 259,435 bytes in 4,154 blocks
==21494==      possibly lost: 117,121 bytes in 172 blocks
==21494==    still reachable: 100,000 bytes in 522 blocks
==21494==         suppressed: 0 bytes in 0 blocks
==21494== Reachable blocks (those to which a pointer was found) are not shown.
==21494== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==21494== 
==21494== For counts of detected and suppressed errors, rerun with: -v
==21494== ERROR SUMMARY: 160 errors from 160 contexts (suppressed: 4 from 3)
```

itdaniher, 2015-02-24:
output of

`valgrind --leak-check=full --show-leak-kinds=all ./pixelpulse2 2>log`

at http://itdaniher.com/p?hwiE

closed 2015-03-05 by itdaniher

schanur, 2015-10-22:
Same problem in my codebase. Did you find a solution to the problem with the 6 open file descriptors? 3 of them are ok to be open at exit(0, 1, 2). The other 3 are not.