analogdevicesinc/Pixelpulse2 #63: voltage/current _are_ continuous

rgetz, 2015-02-25:
When displaying square waves, the voltage/current waveforms don't look continuous. (see attachment)

I get that we are displaying "points", but shouldn't there be an option to connect points?

Thanks
-Robin

[image]

itdaniher, 2015-03-05:
The piecewise definition of a time-domain squarewave suggests no time spent at a voltage other than vmin or vmax. The M1K's output stage is capable of slew rates much faster than the sampling bandwidth, so the device produces a fairly accurate representation of a square wave.

Because of how points are rendered to the screen, I suspect that providing OpenGL with adjacency information between successive datapoints and telling it to "connect the dots" will be implementable, but not without nontrivial modifications to the rendering system. Perhaps @kevinmehall could chime in?

I'd like to see this option implemented by 1.0, but it's not going to make the initial release.

itdaniher, 2015-04-23:
[image]

There's now a renderer implementation (qsgflatcolormaterial branch) using the "connect-the-dots" fallacious-but-simplistic visual representation of sampled signals.

Good reads on potential better representations:
http://www.windytan.com/2014/01/misleading-representations-of-discrete.html
http://www.windytan.com/2013/03/rendering-pcm-with-simulated-phosphor.html

I'll add an option for superimposing connect-the-dots on the dots themselves.

rgetz, 2016-04-01:
I think this is the same as #166 ?