analogdevicesinc/Pixelpulse2 #64: usb device detach crashes pp2 when running on win7

itdaniher, 2015-02-26:
Even with the latest builds, pulling an M1K from a USB port seems to reliably crash Pixelpulse2 on Windows 7. Hopefully easier to diagnose with software traces on crash events.

itdaniher, 2015-03-03:
@dNechita current implementation of tracing at 7d95db246e3d42e4f6dec7ef980fc609d34da22e does not print anything on this crash. Dan, can you take a look at reproducing this crash (as I did on the call this AM) and coerce your code to give a dump?

dNechita, 2015-03-06:
I've got good news and bad news.

The bad news:
Detaching usb device crashes (application hangs) pp2 on linux (Ubuntu 14.04.1) as well but this happens less frequent that on win7: after dozen device removes or more.

The good news:
I used this crash to get the full backtrace using gdb:

Session::detached ser: 2031315434393235323030313330303
m_devices.size() 0
Session::attached ser: 2031315434393235323030313330303
0x7fffe0011110
0x7fffe0011110
device insert 0x7fffe0011110
Session::detached ser: 2031315434393235323030313330303
m_devices.size() 0
Session::attached ser: 2031315434393235323030313330303
0x7fffe00114d0
0x7fffe00114d0
device insert 0x7fffe00114d0
Session::detached ser: 2031315434393235323030313330303
libusb_submit_transfer in -4
libusb_submit_transfer in -4
libusb_submit_transfer in -4
libusb_submit_transfer in -4
libusb_submit_transfer in -4
libusb_submit_transfer in -4
libusb_submit_transfer in -4
libusb_submit_transfer in -4
libusb_submit_transfer out -4
libusb_submit_transfer out -4
libusb_submit_transfer out -4
libusb_submit_transfer out -4
libusb_submit_transfer out -4
libusb_submit_transfer out -4
libusb_submit_transfer out -4
libusb_submit_transfer out -4
^C
Program received signal SIGINT, Interrupt.
pthread_cond_wait@@GLIBC_2.3.2 ()
    at ../nptl/sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
185     ../nptl/sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S: No such file or directory.
(gdb) bt full
#0  pthread_cond_wait@@GLIBC_2.3.2 ()
    at ../nptl/sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
No locals.
#1  0x00007ffff5e8266c in std::condition_variable::wait(std::unique_lock<std::mutex>&) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
No symbol table info available.
#2  0x000000000041ba17 in std::condition_variable::wait<Session::end()::__lambda1>(std::unique_lock<std::mutex> &, Session::__lambda1) (this=0x754718, 
    __lock=..., __p=...) at /usr/include/c++/4.8/condition_variable:93
No locals.
#3  0x000000000041b2bb in Session::end (this=0x7545e0)
    at ../libsmu/session.cpp:211
        lk = {_M_device = 0x7546f0, _M_owns = true}
#4  0x0000000000408945 in SessionItem::onDetached (this=0x7fffffffe440, 
    device=0x7fffe00114d0) at ../SMU.cpp:130
No locals.
#5  0x00000000004152a1 in QtPrivate::FunctorCall<QtPrivate::IndexesList<0>, QtPrivate::List<Device*>, void, void (SessionItem::*)(Device*)>::call (f=
    (void (SessionItem::*)(SessionItem * const, Device *)) 0x40890c <SessionItem::onDetached(Device*)>, o=0x7fffffffe440, arg=0x7fffe00110f0)
    at /usr/include/qt5/QtCore/qobjectdefs_impl.h:508
No locals.
#6  0x0000000000414e29 in QtPrivate::FunctionPointer<void (SessionItem::*)(Devic---Type <return> to continue, or q <return> to quit---
e*)>::call<QtPrivate::List<Device*>, void> (f=
    (void (SessionItem::*)(SessionItem * const, Device *)) 0x40890c <SessionItem::onDetached(Device*)>, o=0x7fffffffe440, arg=0x7fffe00110f0)
    at /usr/include/qt5/QtCore/qobjectdefs_impl.h:527
No locals.
#7  0x0000000000414363 in QtPrivate::QSlotObject<void (SessionItem::*)(Device*), QtPrivate::List<Device*>, void>::impl (which=1, this_=0x750e30, 
    r=0x7fffffffe440, a=0x7fffe00110f0, ret=0x0)
    at /usr/include/qt5/QtCore/qobject_impl.h:149
No locals.
#8  0x00007ffff65ce22e in QObject::event(QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#9  0x00007ffff65a5efd in QCoreApplication::notify(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#10 0x00007ffff65a5c2d in QCoreApplication::notifyInternal(QObject*, QEvent*)
    () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#11 0x00007ffff65a7e07 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#12 0x00007ffff65f2cd3 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
---Type <return> to continue, or q <return> to quit---
No symbol table info available.
#13 0x00007ffff3aa7e04 in g_main_context_dispatch ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#14 0x00007ffff3aa8048 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#15 0x00007ffff3aa80ec in g_main_context_iteration ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#16 0x00007ffff65f298c in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#17 0x00007ffff65a496b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#18 0x00007ffff65ab0e1 in QCoreApplication::exec() ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
No symbol table info available.
#19 0x0000000000406f77 in main (argc=1, argv=0x7fffffffe568) at ../main.cpp:34
        app = <incomplete type>
        engine = <incomplete type>
        smu_session = {<QObject> = {<No data fields>}, 
          static staticMetaObject = {d = {
---Type <return> to continue, or q <return> to quit---
              superdata = 0x7ffff69dbf80 <QObject::staticMetaObject>, 
              stringdata = 0x42f7a0 <qt_meta_stringdata_SessionItem>, 
              data = 0x42fb80 <qt_meta_data_SessionItem>, 
              static_metacall = 0x4216a8 <SessionItem::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)>, relatedMetaObjects = 0x0, 
              extradata = 0x0}}, m_session = 0x7545e0, m_active = true, 
          m_continuous = false, m_sample_rate = 40000, m_sample_count = 4000, 
          m_devices = {{p = {static shared_null = {ref = {atomic = {
                      _q_value = -1}}, alloc = 0, begin = 0, end = 0, array = {
                    0x0}}, d = 0x746230}, d = 0x746230}}}
        versions = {d = 0x746480}
        r = 32767
(gdb)

itdaniher, 2015-03-06:
Looks like we need to replace the use of 'wait' in `m_completion.wait(lk, [&]{ return m_active_devices == 0; });` with http://en.cppreference.com/w/cpp/thread/condition_variable/wait_until, adding timeout handling code to the subsequent calls.

itdaniher, 2015-03-27:
The latest set of logs I have relating to this issue:

control transfer failed with code -4
libusb_submit_transfer in -4
LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED 100
LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED 100
cancel error in: LIBUSB_ERROR_NOT_FOUND out: LIBUSB_ERROR_NOT_FOUND
libusb_submit_transfer in -4
libusb_submit_transfer out -4
libusb_submit_transfer out -4
qml: started
Session::detached ser: 2031315434393235313430313430313
LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED 100
LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED 100
cancel error in: LIBUSB_ERROR_NOT_FOUND out: LIBUSB_ERROR_NOT_FOUND
timed out

-----

The hang happens when start_run submits transfers to a detached device before the onDetached handler is called.

itdaniher, 2015-03-31:
Even at libsmu@b086af4fbbc728597f49ac2d31736ca42820cfac, something is still erroneously unfreed or uncanceled on device detach on Windows:

> LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED 100
> cancel error in: LIBUSB_ERROR_NO_DEVICE out: LIBUSB_ERROR_NO_DEVICE
> timed out

itdaniher, 2015-04-07:
Using a minimal test case (https://github.com/analogdevicesinc/libsmu/commit/bda9eec6594d4c88d021c83023e06be3105b022b) I was able to get the following traceback from a detach event:

```
    Program received signal SIGSEGV, Segmentation fault.
    [Switching to Thread 5608.0x1f88]
    0x004027e4 in __gnu_cxx::__atomic_add (__mem=0x88394, __val=1) at c:/mingw32/i686-w64-    mingw32/include/c++/ext/atomicity.h:53
    53      c:/mingw32/i686-w64-mingw32/include/c++/ext/atomicity.h: No such file or directory.
(gdb) bt
    #0  0x004027e4 in __gnu_cxx::__atomic_add (__mem=0x88394, __val=1) at c:/mingw32/i686-w64-mingw32/include/c++/ext/atomicity.h:53
    #1  0x00402881 in __gnu_cxx::__atomic_add_dispatch (__mem=0x88394, __val=1) at c:/mingw32/i686-w64-mingw32/include/c++/ext/atomicity.h:96
    #2  0x004241bf in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_add_ref_copy (this=0x88390) at c:/mingw32/i686-w64-mingw32/include/c++/bits/shared_ptr_base.h:133
    #3  0x00423de8 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count (this=0x314fc78, __r=...) at c:/mingw32/i686-w64-mingw32/include/c++/bits/shared_ptr_ba
    #4  0x00423ba1 in std::__shared_ptr<Device, (__gnu_cxx::_Lock_policy)2>::__shared_ptr (this=0x314fc74) at c:/mingw32/i686-w64-mingw32/include/c++/bits/shared_ptr_base.h:
    #5  0x00423669 in std::shared_ptr<Device>::shared_ptr (this=0x314fc74) at c:/mingw32/i686-w64-mingw32/include/c++/bits/shared_ptr.h:103
    #6  0x004032d6 in Session::find_existing_device (this=0x733bb8, device=0x738720) at session.cpp:161
    #7  0x00402d94 in Session::detached (this=0x733bb8, device=0x738720) at session.cpp:83
    #8  0x00402e52 in hotplug_callback_usbthread@16 (ctx=0x733c98, device=0x738720, event=LIBUSB_HOTPLUG_EVENT_DEVICE_LEFT, user_data=0x733bb8) at session.cpp:99
    #9  0x00409acd in usbi_hotplug_match_cb (ctx=0x733c98, dev=<optimized out>, event=<optimized out>, hotplug_cb=0x7380f0) at hotplug.c:179
    #10 0x00409b58 in usbi_hotplug_match (ctx=0x733c98, dev=0x738720, event=LIBUSB_HOTPLUG_EVENT_DEVICE_LEFT) at hotplug.c:192
    #11 0x0040a65b in handle_events (ctx=ctx@entry=0x733c98, tv=tv@entry=0x314fdf8) at io.c:2072
    #12 0x0040b028 in libusb_handle_events_timeout_completed@12 (ctx=ctx@entry=0x733c98, tv=tv@entry=0x314fe38, completed=completed@entry=0x0) at io.c:2186
    #13 0x0040b11e in libusb_handle_events@4 (ctx=0x733c98) at io.c:2262
    #14 0x00402e7c in Session::<lambda()>::operator()(void) const (__closure=0x738150) at session.cpp:108
    #15 0x0040492f in std::_Bind_simple<Session::start_usb_thread()::<lambda()>()>::_M_invoke<>(std::_Index_tuple<>) (this=0x738150) at c:/mingw32/i686-w64-mingw32/include/c
    #16 0x004048a6 in std::_Bind_simple<Session::start_usb_thread()::<lambda()>()>::operator()(void) (this=0x738150) at c:/mingw32/i686-w64-mingw32/include/c++/functional:16
    #17 0x00404844 in std::thread::_Impl<std::_Bind_simple<Session::start_usb_thread()::<lambda()>()> >::_M_run(void) (this=0x738144) at c:/mingw32/i686-w64-mingw32/include/
    #18 0x6fcb8c4f in libstdc++-6!execute_native_thread_routine () from c:\Users\IDANIHER\Desktop\distrib-release\libstdc++-6.dll
    #19 0x64944f11 in pthread_create_wrapper () from c:\Users\IDANIHER\Desktop\distrib-release\libwinpthread-1.dll
    #20 0x76641287 in msvcrt!_itow_s () from C:\Windows\syswow64\msvcrt.dll
    #21 0x76641328 in msvcrt!_endthreadex () from C:\Windows\syswow64\msvcrt.dll
    #22 0x760833aa in KERNEL32!BaseThreadInitThunk () from C:\Windows\syswow64\kernel32.dll
    #23 0x7708bf32 in ntdll!RtlInitializeExceptionChain () from C:\Windows\system32\ntdll.dll
    #24 0x7708bf05 in ntdll!RtlInitializeExceptionChain () from C:\Windows\system32\ntdll.dll
    #25 0x00000000 in ?? ()
```

itdaniher, 2015-04-07:
For context:

```
0,0,1.98344
0,1,0.000539311
1,0,1.98093
1,1,0.000104429
0,0,1.98459
0,1,0.000669013
1,0,1.98199
1,1,0.000112058
ITransfer error LIBUSB_TRANSFER_STALL 0x71d084
OTransfer error LIBUSB_TRANSFER_STALL 0x71e75c
OTransfer error LIBUSB_TRANSFER_STALL 0x71ec04
OTransfer error LIBUSB_TRANSFER_STALL 0x71f0ac
OTransfer error LIBUSB_TRANSFER_STALL 0x71f554
OTransfer error LIBUSB_TRANSFER_STALL 0x71f9fc
OTransfer error LIBUSB_TRANSFER_STALL 0x71d964
OTransfer error LIBUSB_TRANSFER_STALL 0x71de0c
OTransfer error LIBUSB_TRANSFER_STALL 0x71e2b4

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 7956.0x2114]
0x004027e4 in __gnu_cxx::__atomic_add (__mem=0x882b4, __val=1) at c:/mingw32/i686-w64-mingw32/include/c++/ext/atomicity.h:53
53      c:/mingw32/i686-w64-mingw32/include/c++/ext/atomicity.h: No such file or directory.
(gdb) bt
```

Notice that the the serial number is never printed - the Session::detached handler never gets that far. Also notice that the transfers stall out, instead of exiting with a no-device-found error.

itdaniher, 2015-04-20:
Okay, Pixelpulse2 no longer crashes on detach, nor does it crash randomly during prolonged runs on Windows. Starting after detach will only return the first chunk of data, so it appears as if there's something broken in the onFinished handler path after replug.

closed 2015-04-24 by itdaniher