analogdevicesinc/Pixelpulse2 #67: input triggering

itdaniher, 2015-02-26:
Input triggering should be added, allowing consistent alignment a measured voltage or current waveform to a DC level or AC zero-crossings. 

This should also render continuous mode streaming usable in conjunction with the scrubbing interface for describing output waveforms.

This may allow source-measure functionality to be extended to waveforms and frequencies, not just DC voltages / currents.

itdaniher, 2015-02-26:
Depending on implementation, this is likely connected to #29 - the addition of zerocrossing detection and period measurement allows phase/frequency explorations.

This is likely a blocker for the completion of the functionality required to close #11 - thoughts, @kevinmehall?

kevinmehall, 2015-02-26:
I see input triggering as a third mode, like continuous in that the device is continually sampling, but using most of the repeated-sweep display code.

And oscope-like "measurements" based on zero-crossing would also be distinct from the frequency-domain plotting in the sidebar based on FFT that I think #29 is referring to.

jonpearson, 2016-04-06:
This would be a great enhancement, and would give students a more real-world representation of what an oscilloscope is.