analogdevicesinc/Pixelpulse2 #74: Lower Drag Dot can be dragged out of the horizontal bounds

dNechita, 2015-03-12:
While drag dots positions are limited on the vertical axis (ranging from 0.0V to 5.0V), on the horizontal axis there is no limit on both left and right edges.
Therefore the dot exceeds both the plot frame and the application window.
This applies to the lower dot of the dot pair which is available for all types of signals except for "Constant" where there is only one available dot movable only on the vertical axis.

Downfalls:
 - the dot overlaps with other GUI elements which damages esthetics.
 - the position of the dot can be lost (not visible anymore) if the dot gets dragged out of the window.

[image]

itdaniher, 2015-03-19:
https://github.com/analogdevicesinc/pixelpulse2/commit/1b327913300b56217cc81b51405bb58c0c59a39d

itdaniher, 2015-03-19:
Resolved this by checking xaxis.width in the onDrag handlers for DragDots.

closed 2015-03-19 by itdaniher