analogdevicesinc/Pixelpulse2 #77: Appveyor Windows version has old dependancy zip file.

G4DCP, 2015-03-16:
Current pixelpulse2.exe requires libcurl & libJannsson-4.

I tried obtaining the missing dll's (maybe I have wrong versions).
  It then crashes when trying to check for an update.

G4DCP, 2015-03-17:
I have tried checking today the dist zip is still missing files for libcurl and libjannsson-4
[image]
The attached image shows what happens when I run the latest exec with the dist zip and what I hope are the correct DLL's. 
Please can someone look at this problem.

dNechita, 2015-03-17:
Yes, I'll take a look at this and try to fix it in the next couple of hours.

Until then, could you please use the 0.1.175 build?
https://ci.appveyor.com/project/analogdevicesinc/pixelpulse2/build/0.1.175/artifacts

Thanks for reporting this and sorry for the inconvenience.

-Dan

G4DCP, 2015-03-17:
Using 0.1.175 is much less likely to crash than the version in the builds area.

Looking forward to the latest version.

dNechita, 2015-03-18:
Things should be working now.

-Dan

closed 2015-03-18 by dNechita

G4DCP, 2015-03-18:
I can confirm that it does work!