analogdevicesinc/Pixelpulse2 #81: Add ability to drag in plot screens

G4DCP, 2015-03-19:
You can drag or zoom with or without shift in the scope views, please can you add drag with and without shift in the plot windows?

G4DCP, 2015-03-21:
Oh ouch!
It has lost the entire message!

In the scope sections of the display you can use shift drag and drag to move the viewpoint.
You can also use to the middle zoom wheel to scale the x axis and shift zoom to scale the y axis and that is very intuitive.
In the plot section of the display which can be optionally turned on zoom and shift zoom work well.
Unfortunately drag and shift drag do not have any effect on the viewpoint. Which is a pity.

Sorry I somehow managed to loose my original post!

itdaniher, 2015-04-10:
Closing as dup of #97 - this has been merged into master!

closed 2015-04-10 by itdaniher