analogdevicesinc/Pixelpulse2 #85: Sample rate seems incorrect

rgetz, 2015-03-20:
Not sure if this is a firmware issue or not, but the 100kSPS seems to actually by 65kSPS.

itdaniher, 2015-03-20:
Fixed this issue at https://github.com/analogdevicesinc/pixelpulse2/commit/264cf0e4fcc4de8c7507faa9abf238ef351a20b4. It's a hack for now, blocked by https://github.com/analogdevicesinc/libsmu/issues/23 and telling Qt how to pull sample period from libsmu, rather than having the apparent sample rate determined by the UI and the backend sample rate determined by libsmu.

closed 2015-03-20 by itdaniher

rgetz, 2015-03-20:
I don't think this is closed/fixed -- [the product page](http://www.analog.com/adalm1000)  describes things as 100kSPS. 

Oscilloscope (100 kSPS), function generator (100 kSPS)

Right now, it is not. Reporting the actual sample rate (62.5kSPS) is not the "fix". (although it's better than misinforming people).

If we want to track it in the libsmu repository - that's fine. Closed as a dup makes more sense (to me).