analogdevicesinc/Pixelpulse2 #92: Windows hotplug should start PixelPulse

rgetz, 2015-03-20:
Refinement of software system to support plugging in M1k device, and have Pixelpulse automatically start.

itdaniher, 2015-03-20:
The creation of a small "pixelpulse2-helper.exe" application using LibSMU for hotplug support & device detection that launches pixelpulse2.exe.

damercer, 2015-03-26:
This should be optional. Someone might be looking to run some other software when plugging in the m1k, such as a custom user written program in C++ or Python.

rgetz, 2015-03-26:
Doug:

Yeah, this would only work when someone was running the helper app. We could make it optional to run a different app if you think that would be better.

itdaniher, 2015-04-28:
I have this sketched out, not super-high priority to implement, but should land before 1.0, maybe sooner.

https://github.com/itdaniher/pp2trayer

Once it's something besides a template, I'll move to @analogdevicesinc.

itdaniher, 2015-10-06:
Cross-platform autolaunch functionality is implemented at https://github.com/analogdevicesinc/Pixelpulse2-Trayer.

closed 2015-10-06 by itdaniher