analogdevicesinc/Pixelpulse2 #96: document OSX .dmg creation

itdaniher, 2015-03-23:
It's something like:

> make clean && make -j10 && sudo /usr/local/Cellar/qt5/5.4.0/bin/macdeployqt pixelpulse2.app -dmg -always-overwrite -verbose=2

itdaniher, 2015-03-23:
@radhermit Could you take a look at this? Trying to get v0.8 built.

hatchetation, 2015-04-10:
I played around with this yesterday on a clean OS X install, was able to get DMG builds working. Pointer to macdeployqt was a big help. My notes are here, hope they help. (Would be happy to rework these into a doc PR if I'm not totally off-base.)

https://gist.github.com/hatchetation/c074b24bdb8795e72c83

One thing I may have left out -- I think I have libusb installed on all my available Macs, so I'm not entirely sure that the library was included in the build. [The QT OSX deployment doc](http://doc.qt.io/qt-5/osx-deployment.html) imply that this may need to be done manually.

itdaniher, 2015-04-10:
Thanks for the notes! I'm glad to hear your experience has been pleasant. A PR improving the README.mkd instructions for OSX would be great. I'm hoping to set up TravisCI and start offering nightly OSX builds, but probably won't get there until after 1.0.

itdaniher, 2015-04-21:
Have some pending work on this at https://travis-ci.org/analogdevicesinc/Pixelpulse2.

itdaniher, 2015-04-21:
See also: https://github.com/analogdevicesinc/Pixelpulse2/blob/travis-osx/.travis.yml

itdaniher, 2015-04-21:
I just emailed Travis-CI to ask about enabling OSX workers for the Pixelpulse2 project. We'll see....

itdaniher, 2015-04-21:
Travis-CI OSX worker is enabled & successfully building PP2. The last few hangups with S3 deploy are hopefully fixed.

itdaniher, 2015-04-21:
OSX Nightly .dmgs are now available at http://pixelpulse2nightly.s3-website-us-east-1.amazonaws.com/pixelpulse2.dmg.

itdaniher, 2015-04-23:
Nightly DMG tested and reported functional.

itdaniher, 2015-04-23:
Link to nightly included in Readme. Travis-CI config file is a list of the relevant commands, but documenting how to build it is less important now that we have autogenerated .dmgs.

closed 2015-04-23 by itdaniher