analogdevicesinc/Pixelpulse2 #103: Missing XY plot for Channel B

dNechita, 2015-03-30:
There's only one XY plot that corresponds to Channel A. 
[image]

@itdaniher can you give me a hand with this?

Thanks,
-Dan

itdaniher, 2015-03-30:
This issue occurs when the xyplot object in https://github.com/analogdevicesinc/Pixelpulse2/blob/master/qml/PlotPane.qml is reset on device detach/attach. The channels are added sequentially, and the first channel (channel A) fills the height of the PlotPane before channel B is added.

The fillheight parameter is set at https://github.com/analogdevicesinc/Pixelpulse2/blob/master/qml/XYPlot.qml#L9. If you comment out that line, and add a property "height:320" the visual glitching ends. I'm working on a dynamic fix without a predefined height value.

itdaniher, 2015-03-30:
That should do it. Cheers!

commit ec68b1bb (2015-03-30): closes #103 - keep xy subplot height tied to window height
closed 2015-03-30 by itdaniher