analogdevicesinc/Pixelpulse2 #106: optionally increase waveform brightness

itdaniher, 2015-03-31:
On mediocre screens, it can be difficult to view the waveform at high zoom levels.

kevinmehall, 2015-03-31:
When the [point size](https://github.com/analogdevicesinc/Pixelpulse2/blob/ce8c213f489859825b36bb55778f8e0fb7fe741a/qml/SignalRow.qml#L321) hits its maximum, it should start increasing the brightness of the color as you zoom in further.

rgetz, 2016-03-23:
Sometimes - (on high resolution screens) - it's nearly impossible to see a line one pixel wide. 1080p is fine with the existing setup - new screens are 3840 x 2160 - and it's impossible to see...

Updating to bug.

luciansin, 2016-04-05:
It happened also to me and I agree with Robin. This is clearly a bug, not an enhancement.

jonpearson, 2016-04-06:
While not a show-stopper for the labs, this is important in the overall operation.  It would also help when projecting the PixelPulse screen, as we've had difficulty seeing the traces when using a projector.

AlexandraTrifan, 2016-05-05:
The feature added for issue #176 allows users to manually change the brightness for plots and for the phosphor dot. Should the brightness increase automatically as you zoom in, or is this alright?

rgetz, 2016-05-10:
Feature #176 fixes this - closing.

closed 2016-05-10 by rgetz