analogdevicesinc/Pixelpulse2 #115: firmware auto-update

itdaniher, 2015-04-08:
on platforms where it's possible, Pixelpulse2 should be capable of automatically fetching and updating new device firmware.

itdaniher, 2015-04-20:
I believe the best path for cross-platform support is to ship bossac.exe from https://github.com/shumatech/BOSSA with Windows installers of Pixelpulse2.exe. The project is BSD licensed, bossac is tiny / standalone. Writing a libbossa Qt wrapper is definitely, but probably more work than it's worth.

JS/C++ ports of [sam-ba.py](https://github.com/analogdevicesinc/m1k-fw/blob/master/scripts/sam-ba.py) will not work on Windows as the serial port driver is automatically bound and is difficult to unbind.

itdaniher, 2015-09-29:
https://ci.appveyor.com/project/analogdevicesinc/bossa

https://travis-ci.org/analogdevicesinc/BOSSA

http://doc.qt.io/qt-5/qprocess.html

itdaniher, 2015-09-29:
bossac was tested and confirmed to work on OSX, Windows, and Linux today.

itdaniher, 2015-10-06:
Remaining todo here is to set up releases on https://github.com/analogdevicesinc/m1k-fw and write some Javascript to grab the latest m1000.bin release, save to a [temporary file](http://doc.qt.io/qt-5/qtemporaryfile.html), call `session.devices[0].ctrl_transfer(0xbb, 0, 0)` to trigger device erase / DFU enumeration, and call BossacWrapper.flashByFilename(tempFile.fileName());

itdaniher, 2015-10-06:
@dbogdan - shall we tag 2.02?
@rgetz - I'm assuming we emphatically do *not* want this to do the fetch & flash without user approval?
@radhermit - besides now-functional TLS in PP2 and using Github-hosted releases, are there additional security measures you believe would be beneficial?

dbogdan, 2015-10-07:
@itdaniher, yes, we can tag 2.02.

itdaniher, 2015-10-12:
@dbogdan v2.02 tagged.

acozma, 2015-12-04:
Firmware auto update is available in Pixelpulse v0.86. 
More details on the firmware update procedure form Pixelpulse can be found here: https://wiki.analog.com/university/tools/m1k-firmware-upgrade

closed 2015-12-04 by acozma