analogdevicesinc/Pixelpulse2 #116: ChromeOS Support

itdaniher, 2015-04-08:
I've made initial forays into building the relevant Qt libraries with the Native Client toolchain for ChromeOS. This should be doable, but the documentation on the toolchains is both partial and partially outdated. 

See http://wiki.qt.io/Qt_for_Google_Native_Client, https://github.com/msorvig/qt5-qtdeclarative-nacl, https://github.com/msorvig/qt5-qttools-nacl.

itdaniher, 2015-04-09:
"The Qt for NaCl project has quietly received an update and we are now opening up the repo. This is a dev branch (not a tested/packaged/supported release), with the standard LGPL3/GPL2+/Commercial licensing.

The main repos are qt5-qtbase-nacl and qt5-qtdeclarative-nacl, branch "nacl-5.4" on both. This qt5-qttools-nacl repo is now obsolete since nacldeployqt is provided by qtbase. See qtbase/nacl-readme for install instructions, which should get you up and running with a simple Qt Quick example." @radhermit

radhermit, 2015-04-09:
Hah, nice timing. That seems more helpful now then when I looked a week or two ago.

itdaniher, 2015-04-16:
The Toshiba Chromebook2 ($300) works well with the M1K and Pixelpulse2 in a "crouton" (https://github.com/dnschneid/crouton) Debian testing / x-in-window chroot container.

I'm pleasantly surprised.

jonpearson, 2016-04-06:
No problem for the labs, but should be done going forward.