analogdevicesinc/Pixelpulse2 #123: Print CVS data with fixed field length

UweBonnes, 2015-04-20:
Hello,

for visual inspection it would be usefull to print the CVS data with constant fieldwidth. Otherwise a ragged output is generated, making it harder for the eye to follow.

itdaniher, 2015-04-20:
I'll see what I can do! For visual inspection of data from the M1k, might I recommend Pixelpulse2? ;)

UweBonnes, 2015-04-20:
When pixelview can also zoom vertical, the CVS constant filedwith feature is no longer needed :-)

itdaniher, 2015-04-20:
It can! Two ways.

* Hold shift and use your mouse's scroll functionality.
* Right click and drag on the Y-Axes labels.

This exchange has remotivated me to update the UI interactions... Hope those two solutions work!

itdaniher, 2015-04-20:
I made some quick modifications to dataexport.js -

[image]

The issue is complicated with the presence of negative signs. I could write a more complicated padding algorithm (besides just calling toFixed and adding a tab) but if Y-axis zoom was the motivation here, then I'm not going to push on the issue.

itdaniher, 2015-04-20:
Fixed at 7a0d50a

closed 2015-04-20 by itdaniher

UweBonnes, 2015-04-21:
Ian> Closed #123.

I propose to print at least 6 digit and for current where maximum range is
0.2 A 7 digits or perhaps print the current in +/-xxx.xx mA. Otherwise we
cut of ADC resolution.
The CVS line indicating the meaning of the values printed should also be
somehow distingushable from the data. E.g. gnuplot ignores lines starting
with hash '#'.
What about?

diff --git a/qml/dataexport.js b/qml/dataexport.js
index 8d87802..64bf23d 100644
--- a/qml/dataexport.js
+++ b/qml/dataexport.js
@@ -5,7 +5,7 @@ var dumpsample = function (columns, labels) {
       throw("label length mismatches number of columns");
    }
    var lengths = columns.map(function(x) {return x.length;})
-   var csvContent = '';
+   var csvContent = '#';
    for (var i = 0; i < labels.length; i++) {
        csvContent += labels[i] + ((i != (labels.length-1)) ? "," : "");
    }
@@ -13,8 +13,8 @@ var dumpsample = function (columns, labels) {
    var minimumLength = Math.min.apply(null, lengths);
    for (var i = 0; i < minimumLength; i++) {
        for (var j = 0; j < columns.length; j++) {
-            var x = columns[j][i].toFixed(4);
-            x = (x == 0) ? (0.00001).toFixed(4) : x;
+            var x = columns[j][i].toFixed(6);
+            x = (x == 0) ? (0.00001).toFixed(6) : x;
             csvContent += (x >= 0 ? "+" : "") + x + ((j != (columns.length-1)) ? "," : "");
         }
         csvContent += (i != (minimumLength-1) ? "\n" : "");
Uwe Bonnes                bon@elektron.ikp.physik.tu-darmstadt.de

Institut fuer Kernphysik  Schlossgartenstrasse 9  64289 Darmstadt
--------- Tel. 06151 162516 -------- Fax. 06151 164321 ----------