analogdevicesinc/Pixelpulse2 #128: Pixelpulse2 fails to run when Qt 5.4 is present on the machine

dNechita, 2015-04-22:
Pixelpulse2 outputs the following on a windows 7 machine that has Qt 5.4 installed on:
[image]

Hint: I can make Pixelpulse2 to run if I change the path of the installation of Qt (by changing the Qt folder's name).

itdaniher, 2015-04-22:
http://stackoverflow.com/questions/16773789/qt5-static-build-yields-failed-to-load-platform-plugin-windows/17271172#17271172

itdaniher, 2015-04-22:
> QCoreApplication::addLibraryPath("./");

itdaniher, 2015-04-22:
With the latest build, ["Dependency Walker](www.dependencywalker.com) shows the following for Pixelpulse2.exe:

[image]

@dNechita Would you be so kind as to run depends.exe on Pixelpulse2 without renaming your Qt folder?

dNechita, 2015-04-23:
[image]

I'll send you an email with the dependency walker image.

itdaniher, 2015-04-23:
I have reproduced this issue on my Windows 8 testing laptop. Issue confirmed.

closed 2015-04-27 by itdaniher

dNechita, 2015-04-28:
I can confirm that the issue is fixed and I no longer have this problem on my machine.

Thanks!

commit 7c470342 (2015-04-27): Merge pull request #132 from analogdevicesinc/fix-#128

Fix #128