analogdevicesinc/Pixelpulse2 #137: The check of the latest git release fails every time

dNechita, 2015-05-07:
[image]

itdaniher, 2015-05-07:
This looks as if there's a network connectivity problem. Could you get a screenshot of the "About" pane and make sure you can ping github.com?

dNechita, 2015-05-08:
[image]
[image]

Also the C implementation of phone home that pixelpulse2 used in the past is working for me. Therefore it doesn't look like a connectivity problem.

closed 2015-05-30 by itdaniher