analogdevicesinc/Pixelpulse2 #140: QOpenGLShaderProgram

vthielen, 2015-05-19:
I downloaded and installed the windows version of the 0.85 version of pixel pulse2 (called pixelpulse2-v0.85_win_setup.exe).
But when I  try to run it, I first get a 4 messages "QSslSocket: cannot resolve..." and then the message in the window below starts scrolling across the screen.
There is another window that appears as well (the graphics window, I presume), but that stays blank.
I am using A PC running windows 7
I downloaded and installed QT5, but that didn't help.

[image]

itdaniher, 2015-05-30:
Hi there, sorry to hear of your issues. Could you let me know more information about your computer? It sounds like your graphics card may not have all the drivers it needs.

Thanks,
Ian

damercer, 2015-06-01:
I have this same error message on an older Dell desk-top. I think this issue happens because the computer has an older graphics chip set. The newer OpenGL drivers won't or don't support these older
graphics chips. This Q&A thread may provide a solution:

https://forum.qt.io/topic/52625/qopenglshaderprogram-could-not-create-shader-program

Doug

acozma, 2015-12-04:
The issues is due to OpenGL not being supported by older graphics cards. Changing this issue to 'enhancement', to investigate if Pixelpulse can fallback to DirectX or something else when OpenGL is not supported.

rgetz, 2015-12-04:
Qt should fall back to a software render when GL isn't available. (My understanding)

vthielen, 2015-12-07:
Hello,

Thanks for the reply.
I installed the driver separately, and now it works.
I still have 2 problems though:

- when I click on the settings icon (on the top left), the menu appears,
but dissapears after one second.
When I move the mouse it sometimes reappears (blinks), and it's possible to
select some things, but the sampletime submenu doesn't appear

- on channel 2 there is a hardware instability, that looks like an
oscillation on the current (and to a smaller extent also on the voltage)
when nothing is connected. (see attached file)
Maybe a loose decoupling cap?
Any idea?

Kind Regards,

Bart Van Thielen

acozma, 2015-12-10:
Can you please upgrade Pixelpulse to the latest version? https://github.com/analogdevicesinc/Pixelpulse2/releases/tag/v0.87

Also please update the firmware on your device:
https://wiki.analog.com/university/tools/m1k-firmware-upgrade

Once you have everything up to date can you please try again and let us know if there's any improvement?

Thanks,
Andrei

closed 2019-07-03 by AlexandraTrifan