analogdevicesinc/Pixelpulse2 #141: Pixelpulse2: crash-report

Paulibaer, 2015-05-30:
I use openSUSE 13.2, KDE4 and QT5.
Pixelpulse2 is running, but if I try to use "export data", "restore session" or "restore session" the program crashed with the errormessage:

 QWidget: Cannot create a QWidget without QApplication

Can anybody help me to solve the problem?

Karl Brandt

itdaniher, 2015-05-30:
Hey - I'll check into this. Could you confirm the version of Qt5 you're using, ie 5.4.1?

Thanks,
Ian

Paulibaer, 2015-05-30:
Hello itdantiher,

yes, it is QT5 Verision 5.4.1-4.1

Thanks
Karl Brandt

itdaniher, 2015-06-01:
How did you install Qt? Repositories, or from the official binary package? I don't have an openSUSE install, but running Pixelpulse2 with Qt5.4.1 on Fedora, Debian, and Ubuntu, I'm unable to reproduce that message with those menu items.

Paulibaer, 2015-06-01:
Hallo Ian,

yesterday I installed pixelpulse2 on another Laptop with 
QT5.4.1. (openSUSE)

Pixelpulse2 works without problems.

The difference between Laptop- and Desktop-Installations 
(for example):

-libQt5Core5-5.4.1-199.1.x86_64.rpm
+libQt5Core5-5.4.1-4.2.x86_64.rpm

It is impossible (for me) to (re)install the same Libs on 
Desktop and Laptop.
The Installer (YAST2) lists a lot of dependencies and I cant 
solve them :-(

I will do my work with my laptop and send the data to the 
desktop for evaluation and plotting.

I hope, that the next update for openSuSE solve the problem.

I have another question for pixelpulse2.x:

is it possible to integrate a trigger?
e.g the measure on channel a and/or b starts by xV (x = 
variable)

Thanks for your assistance
Karl Brandt

damercer, 2015-06-01:
I have this same error message on an older Dell desk-top. I think this issue happens because the computer has an older graphics chip set. The newer OpenGL drivers won't or don't support these older
graphics chips. This Q&A thread may provide a solution:

https://forum.qt.io/topic/52625/qopenglshaderprogram-could-not-create-shader-program

Doug

itdaniher, 2015-06-03:
Hi Karl, adding a trigger is on the todo list for Pixelpulse2, it should make the next version. I'm glad you were able to get an install that works, please update me when you have a chance to upgrade OpenSUSE.

Paulibaer, 2015-06-11:
Hi Ian,

my problems with the qt-lib is "solved".

Pixelpulse2 works fine with XFCE but not with KDE 4.x on the 
same PC and installed qt-libs.  It seems, that the installed 
kde-libs (some of them are KDE5-libs) not compatible :-(

Karl Brandt

itdaniher, 2015-06-11:
*Weird!*

Thanks for the update. I've tested Pixelpulse2 on Kubuntu 15.04, glad your issues were confined to a particular environment/version combination. Closing the issue.

closed 2015-06-11 by itdaniher