analogdevicesinc/Pixelpulse2 #142: Crash on startup, OSX 10.9.5

wb1000000, 2015-06-08:
Pixelpulse 0.85 and 0.8 crash on startup on my Mac (2013 RMBP 13"). The pre-release OSX version works. The error occurs with and without the ADALM1000 attached. The crash-report:

Process:         Pixelpulse2 [2502]
Path:            /Applications/pixelpulse2.app/Contents/MacOS/Pixelpulse2
Identifier:      com.yourcompany.Pixelpulse2
Version:         ???
Code Type:       X86-64 (Native)
Parent Process:  launchd [242]
Responsible:     Pixelpulse2 [2502]
User ID:         501

Date/Time:       2015-06-08 19:44:23.104 +0200
OS Version:      Mac OS X 10.9.5 (13F1077)
Report Version:  11
Anonymous UUID:  681BD06D-EC4E-1C3A-E512-D68641472026

Sleep/Wake UUID: 0625C714-27C1-4E61-89A1-C2C93E1633F1

Crashed Thread:  0  Dispatch queue: com.apple.main-thread

Exception Type:  EXC_CRASH (SIGABRT)
Exception Codes: 0x0000000000000000, 0x0000000000000000

Application Specific Information:
abort() called

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff8f505866 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff8f24835c pthread_kill + 92
2   libsystem_c.dylib             	0x00007fff8cf87b1a abort + 125
3   org.qt-project.QtCore         	0x0000000104b15d29 0x104af3000 + 142633
4   org.qt-project.QtCore         	0x0000000104b171e1 QMessageLogger::fatal(char const*, ...) const + 161
5   org.qt-project.QtGui          	0x00000001045cb0b7 QGuiApplicationPrivate::createPlatformIntegration() + 6359
6   org.qt-project.QtGui          	0x00000001045cb0db QGuiApplicationPrivate::createEventDispatcher() + 27
7   org.qt-project.QtCore         	0x0000000104d308e1 QCoreApplication::init() + 113
8   org.qt-project.QtCore         	0x0000000104d30857 QCoreApplication::QCoreApplication(QCoreApplicationPrivate&) + 39
9   org.qt-project.QtGui          	0x00000001045c8504 QGuiApplication::QGuiApplication(int&, char**, int) + 228
10  com.yourcompany.Pixelpulse2   	0x00000001041c1582 main + 162
11  com.yourcompany.Pixelpulse2   	0x00000001041c1164 start + 52

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff8f505e6a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8f248f08 _pthread_wqthread + 330
2   libsystem_pthread.dylib       	0x00007fff8f24bfb9 start_wqthread + 13

Thread 2:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff8f506662 kevent64 + 10
1   libdispatch.dylib             	0x00007fff8f583421 _dispatch_mgr_invoke + 239
2   libdispatch.dylib             	0x00007fff8f583136 _dispatch_mgr_thread + 52

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff8f505e6a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8f248f08 _pthread_wqthread + 330
2   libsystem_pthread.dylib       	0x00007fff8f24bfb9 start_wqthread + 13

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x00007fff78bea310  rcx: 0x00007fff5ba437b8  rdx: 0x0000000000000000
  rdi: 0x0000000000000707  rsi: 0x0000000000000006  rbp: 0x00007fff5ba437e0  rsp: 0x00007fff5ba437b8
   r8: 0x0000000000000040   r9: 0x00007fff5ba43700  r10: 0x0000000008000000  r11: 0x0000000000000206
  r12: 0x0000000104e27b88  r13: 0x00007f867a704220  r14: 0x0000000000000006  r15: 0x00007fff5ba43cc0
  rip: 0x00007fff8f505866  rfl: 0x0000000000000206  cr2: 0x0000000104b97a20
  
Logical CPU:     0
Error Code:      0x02000148
Trap Number:     133

Binary Images:
       0x1041bc000 -        0x1041fcfff +com.yourcompany.Pixelpulse2 (???) <B588A736-98A4-3A42-BBB9-3E669B727F5E> /Applications/pixelpulse2.app/Contents/MacOS/Pixelpulse2
       0x10421d000 -        0x104507ff7 +org.qt-project.QtQuick (5.4 - 5.4.1) <677987C8-6DF2-36D2-AFDE-8E1AD311C001> /Applications/pixelpulse2.app/Contents/Frameworks/QtQuick.framework/Versions/5/QtQuick
       0x1045ab000 -        0x104a1dff7 +org.qt-project.QtGui (5.4 - 5.4.1) <EB0D4FE9-8575-338F-84E9-6B6BB14BFE98> /Applications/pixelpulse2.app/Contents/Frameworks/QtGui.framework/Versions/5/QtGui
[... 188 lines cut ...]
__IMAGE                             528K
__LINKEDIT                         69.6M
__TEXT                            109.5M
__UNICODE                           544K
mapped file                        22.4M
shared memory                         4K
===========                      =======
TOTAL                             320.1M

Model: MacBookPro10,2, BootROM MBP102.0106.B03, 2 processors, Intel Core i5, 2.5 GHz, 8 GB, SMC 2.6f59
Graphics: Intel HD Graphics 4000, Intel HD Graphics 4000, Built-In
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, 0x802C, 0x384B54463531323634485A2D314736453120
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x802C, 0x384B54463531323634485A2D314736453120
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x10F), Broadcom BCM43xx 1.0 (5.106.98.100.22)
Bluetooth: Version 4.2.7f4 15112, 3 services, 23 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM256E, 251 GB
USB Device: Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: Hub
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 23.4

itdaniher, 2015-06-09:
@wb1000000 Sorry to hear of your issues. I'll investigate further. I have received multiple reports that the 0.85 image works, but the versions were built using different tools (by hand, vs via travis-ci).

wb1000000, 2015-06-10:
Hi Ian,

If you have some build instruction posted somewhere I can also try building
by hand.
Also if you need some more debugging info let me know.

Best regards,

Jasper

itdaniher, 2015-06-10:
Jasper - If you're feeling brave, you could try following along with the protocol we use for building with travis-ci.

The full configuration is at https://github.com/analogdevicesinc/Pixelpulse2/blob/master/.travis.yml.

This can be abbreviated as:

``` bash
brew update
brew install --build-from-source libusb
brew install qt5
brew linkapps qt5
git clone --recursive https://github.com/analogdevicesinc/Pixelpulse2
cd Pixelpulse2
/usr/local/opt/qt5/bin/qmake
make
/usr/local/opt/qt5/bin/macdeployqt pixelpulse2.app -dmg -always-overwrite -verbose=2 -qmldir=qml
```

Note that this set of instructions requires ["homebrew"](https://github.com/Homebrew/homebrew/) the OSX package manager. 

After you type "make" you should get a pixelpulse2 executable, you may be able to run it from inside the "Pixelpulse2" folder by running "sudo ./pixelpulse2" - if this locally built copy works, then there's something quite strange afoot. :)

Regards,
Ian

wb1000000, 2015-06-10:
Hi Ian,

I followed your instruction above and the build went without a hitch. The program still crashes though, apparently as you already excepted.
Let me know if there is anything else I can try.

Best regards,

Jasper

itdaniher, 2015-06-11:
Perhaps an idle question, but have you tried launching Pixelpulse2 as a root or administrator user?

itdaniher, 2015-06-11:
I also found a similar issue [on stackoverflow](http://stackoverflow.com/questions/21514268/qt-macdeployqt-crash-after-using) - do you perchance have multiple versions of Qt installed? Have you tried running ```./pixelpulse2``` instead of launching via the app?

wb1000000, 2015-06-11:
Hi Ian,

I cleaned all the code and did a rebuild but I skipped the last step:

/usr/local/opt/qt5/bin/macdeployqt pixelpulse2.app -dmg -always-overwrite -verbose=2 -qmldir=qml

I can then run the resulting executable (which is already an .app package), both the one inside the package from the terminal and by double clicking. Does not matter if I'm root or not. It could very well be that I have multiple versions of Qt installed, don’t know how to check that really…

Anyway all is good now. It even recognised my old CEE :-)

Best regards,

Jasper

itdaniher, 2015-06-11:
Hey Jasper - 

Just to confirm, by leaving off the last step, you have a working Pixelpulse2.app?

Thanks,
Ian

wb1000000, 2015-06-11:
Correct, I’ve done:

rm -rf Pixelpulse2/
git clone --recursive https://github.com/analogdevicesinc/Pixelpulse2
cd Pixelpulse2
/usr/local/opt/qt5/bin/qmake
make

Then it works. Once I do:

/usr/local/opt/qt5/bin/macdeployqt pixelpulse2.app -dmg -always-overwrite -verbose=2

Both the app in the .dmg and the .app in the Pixelpulse2 folder break…

Best regards,

Jasper

pettefar, 2015-08-14:
I have the same problem.  When I run it in a terminal it shows: This application failed to start because it could not find or load the Qt platform plugin "cocoa".

Reinstalling the application may fix this problem.
Abort trap: 6

I tried reinstalling of course but to no avail.

Is there a working version somewhere?  Can the OP load his working version somewhere here please?

Cheers,

Nick

rgetz, 2015-08-14:
Pettefar:

were you building from source - or using the release?

https://github.com/analogdevicesinc/Pixelpulse2/releases

?

Thanks
-Robin

pettefar, 2015-08-14:
Just using the release.  I am so naive.
Yosemite 10.10.4

Cheers,

Nick

dNechita, 2015-08-14:
Hi @radhermit 

I'm able to replicate the issue on a Mac with OS X version 10.10.3.
If it helps I can continue to test the application once you add fix patches to it since this Mac is rather "clean" (no extra programs, libraries, tools, etc. installed) or connect remotely to it. Let me know if you need anything.

I've just downloaded the .dmg file and double-clicked on it. Once I run it, I get a popup saying that "pixelpulse2 quit unexpectedly" and the crash report goes like this:

Process:               Pixelpulse2 [695]
Path:                  /Volumes/VOLUME/pixelpulse2.app/Contents/MacOS/Pixelpulse2
Identifier:            com.yourcompany.Pixelpulse2
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Pixelpulse2 [695]
User ID:               501

Date/Time:             2015-08-14 14:04:39.214 +0300
OS Version:            Mac OS X 10.10.3 (14D136)
Report Version:        11
Anonymous UUID:        74BA4B95-4591-7B58-7353-7E54C0CC4EE6

Time Awake Since Boot: 2100 seconds

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Application Specific Information:
abort() called

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff9062b286 __pthread_kill + 10
1   libsystem_c.dylib             	0x00007fff8697cb53 abort + 129
2   org.qt-project.QtCore         	0x00000001033dad29 0x1033b8000 + 142633
3   org.qt-project.QtCore         	0x00000001033dc1e1 QMessageLogger::fatal(char const*, ...) const + 161
4   org.qt-project.QtGui          	0x0000000102e900b7 QGuiApplicationPrivate::createPlatformIntegration() + 6359
5   org.qt-project.QtGui          	0x0000000102e900db QGuiApplicationPrivate::createEventDispatcher() + 27
6   org.qt-project.QtCore         	0x00000001035f58e1 QCoreApplication::init() + 113
7   org.qt-project.QtCore         	0x00000001035f5857 QCoreApplication::QCoreApplication(QCoreApplicationPrivate&) + 39
8   org.qt-project.QtGui          	0x0000000102e8d504 QGuiApplication::QGuiApplication(int&, char**, int) + 228
9   com.yourcompany.Pixelpulse2   	0x0000000102a7d582 main + 162
10  com.yourcompany.Pixelpulse2   	0x0000000102a7d164 start + 52

Thread 1:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff9062c232 kevent64 + 10
1   libdispatch.dylib             	0x00007fff89935a6a _dispatch_mgr_thread + 52

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff9062b94a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8ec6f40d start_wqthread + 13

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff9062b94a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8ec6f40d start_wqthread + 13

Thread 4:
0   libsystem_kernel.dylib        	0x00007fff9062b94a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8ec6f40d start_wqthread + 13

Thread 5:
0   libsystem_kernel.dylib        	0x00007fff9062b94a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8ec6f40d start_wqthread + 13

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x0000000000000006  rcx: 0x00007fff5d187758  rdx: 0x0000000000000000
  rdi: 0x000000000000130f  rsi: 0x0000000000000006  rbp: 0x00007fff5d187780  rsp: 0x00007fff5d187758
   r8: 0x00007fff5d1876c0   r9: 0x00007fff7750d300  r10: 0x0000000008000000  r11: 0x0000000000000206
  r12: 0x00000001036ecb88  r13: 0x00007fd2eac2a690  r14: 0x00007fff7750d300  r15: 0x00007fff5d187c60
  rip: 0x00007fff9062b286  rfl: 0x0000000000000206  cr2: 0x00007fff7641dfd8
[... 217 lines cut ...]
__IMAGE                             528K
__LINKEDIT                         74.2M
__TEXT                            123.0M
__UNICODE                           552K
mapped file                        23.3M
shared memory                         4K
===========                      =======
TOTAL                             346.8M

Model: Macmini7,1, BootROM MM71.0220.B01, 2 processors, Intel Core i5, 1.4 GHz, 4 GB, SMC 2.24f32
Graphics: Intel HD Graphics 5000, Intel HD Graphics 5000, Built-In
Memory Module: BANK 0/DIMM0, 2 GB, DDR3, 1600 MHz, 0x80CE, 0x4B3445384533303445452D45474345000000
Memory Module: BANK 1/DIMM0, 2 GB, DDR3, 1600 MHz, 0x80CE, 0x4B3445384533303445452D45474345000000
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x13B), Broadcom BCM43xx 1.0 (7.15.166.24.3)
Bluetooth: Version 4.3.4f4 15601, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: APPLE HDD HTS545050A7E362, 500,11 GB
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: IR Receiver
USB Device: Microsoft Basic Optical Mouse v2.0
USB Device: Wired Keyboard 600
Thunderbolt Bus: Mac mini, Apple Inc., 26.1
 
-Dan

radhermit, 2015-08-14:
On 2015-08-14 07:15, Dan Nechita wrote:
> Hi @radhermit 

> I'm able to replicate the issue on a Mac with OS X version 10.10.3.
> If it helps I can continue to test the application once you add fix
> patches to it since this Mac is rather "clean" (no extra programs,
> libraries, tools, etc. installed) or connect remotely to it. Let me
> know if you need anything.

> I've just downloaded the .dmg file and double-clicked on it. Once I
> run it, I get a popup saying that "pixelpulse2 quit unexpectedly" and
> the crash report goes like this:

As mentioned by Ian, this feels like a build environment issue if the
pre-release I made manually works fine but the new releases made on
travis-ci crash.

I'm pretty sure I'll be unable to reproduce it if I rebuild it on the
machine here locally.

It might help if we could get a listing of the bundled libraries and
their versions in the crashing app instances.

pettefar, 2015-08-14:
Let me know if you want me to do anything.

Cheers,

Nick

pettefar, 2015-08-14:
Process:               Pixelpulse2 [15749]
Path:                  /Applications/pixelpulse2.app/Contents/MacOS/Pixelpulse2
Identifier:            Pixelpulse2
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        bash [86735]
Responsible:           Terminal [17489]
User ID:               100

Date/Time:             2015-08-15 00:25:18.042 +0100
OS Version:            Mac OS X 10.10.4 (14E46)
Report Version:        11
Anonymous UUID:        1B72BD2B-FB45-42FA-5FD9-1FBADD3F957B

Sleep/Wake UUID:       25567032-D9D6-4254-893D-24D00C6155A4

Time Awake Since Boot: 1900000 seconds
Time Since Wake:       1700000 seconds

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Application Specific Information:
abort() called

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff880c7286 __pthread_kill + 10
1   libsystem_c.dylib             	0x00007fff88fd2b53 abort + 129
2   org.qt-project.QtCore         	0x000000010b6c5d29 0x10b6a3000 + 142633
3   org.qt-project.QtCore         	0x000000010b6c71e1 QMessageLogger::fatal(char const*, ...) const + 161
4   org.qt-project.QtGui          	0x000000010b17b0b7 QGuiApplicationPrivate::createPlatformIntegration() + 6359
5   org.qt-project.QtGui          	0x000000010b17b0db QGuiApplicationPrivate::createEventDispatcher() + 27
6   org.qt-project.QtCore         	0x000000010b8e08e1 QCoreApplication::init() + 113
7   org.qt-project.QtCore         	0x000000010b8e0857 QCoreApplication::QCoreApplication(QCoreApplicationPrivate&) + 39
8   org.qt-project.QtGui          	0x000000010b178504 QGuiApplication::QGuiApplication(int&, char**, int) + 228
9   com.yourcompany.Pixelpulse2   	0x000000010ad6d582 main + 162
10  com.yourcompany.Pixelpulse2   	0x000000010ad6d164 start + 52

Thread 1:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff880c8232 kevent64 + 10
1   libdispatch.dylib             	0x00007fff86e7ba6a _dispatch_mgr_thread + 52

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff880c794a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8577a40d start_wqthread + 13

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff880c794a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8577a40d start_wqthread + 13

Thread 4:
0   libsystem_kernel.dylib        	0x00007fff880c794a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8577a40d start_wqthread + 13

Thread 5:

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x0000000000000006  rcx: 0x00007fff54e97558  rdx: 0x0000000000000000
  rdi: 0x0000000000000513  rsi: 0x0000000000000006  rbp: 0x00007fff54e97580  rsp: 0x00007fff54e97558
   r8: 0x00007fff54e974c0   r9: 0x00007fff756d9300  r10: 0x0000000008000000  r11: 0x0000000000000206
  r12: 0x000000010b9d7b88  r13: 0x00007fc591613830  r14: 0x00007fff756d9300  r15: 0x00007fff54e97a60
  rip: 0x00007fff880c7286  rfl: 0x0000000000000206  cr2: 0x00007fff75f58fd8
  
Logical CPU:     0
Error Code:      0x02000148
Trap Number:     133

Binary Images:
[... 196 lines cut ...]

VM Region Summary:
ReadOnly portion of Libraries: Total=198.8M resident=246.1M(124%) swapped_out_or_unallocated=16777216.0T(8849153064960%)
Writable regions: Total=55.4M written=1432K(3%) resident=1700K(3%) swapped_out=0K(0%) unallocated=53.7M(97%)
 
REGION TYPE                      VIRTUAL
===========                      =======
Dispatch continuations             8192K
Kernel Alloc Once                     4K
MALLOC                             36.3M
MALLOC (admin)                       32K
STACK GUARD                        56.0M
Stack                              9828K
Stack (reserved)                    520K        reserved VM address space (unallocated)
VM_ALLOCATE                         340K
__DATA                             14.3M
__IMAGE                             528K
__LINKEDIT                         74.3M
__TEXT                            124.5M
__UNICODE                           552K
mapped file                        23.3M
shared memory                         4K
===========                      =======
TOTAL                             348.4M
TOTAL, minus reserved VM space    347.8M

wb1000000, 2015-08-15:
Hi,

I've uploaded my OSX build here:

https://github.com/wb1000000/Pixelpulse2/releases/tag/0.85

Maybe it works for you guys.

Best regards,

Jasper

pettefar, 2015-08-15:
./Pixelpulse2 
dyld: Library not loaded: /usr/local/opt/qt5/lib/QtQuick.framework/Versions/5/QtQuick
  Referenced from: /Applications/Pixelpulse2-1.app/Contents/MacOS/./Pixelpulse2
  Reason: image not found
Trace/BPT trap: 5

wb1000000, 2015-08-16:
OK so that one does not work then. Then I think the only solution at the moment is to build from source...
It's not that hard, as Ian posted before you can do:

brew update
brew install --build-from-source libusb
brew install qt5
brew linkapps qt5
git clone --recursive https://github.com/analogdevicesinc/Pixelpulse2
cd Pixelpulse2
/usr/local/opt/qt5/bin/qmake
make

pettefar, 2015-08-16:
Ah yes, that works - thanks for the advice!

On 16 August 2015 at 19:04, wb1000000 <notifications@github.com> wrote:

> brew update
> brew install --build-from-source libusb
> brew install qt5
> brew linkapps qt5
> git clone --recursive https://github.com/analogdevicesinc/Pixelpulse2
> cd Pixelpulse2
> /usr/local/opt/qt5/bin/qmake
> make
>
Regards,

Nick Pettefar

pettefar, 2015-08-16:
Ah yes, that works - thanks for the advice!

(Had to install brew first though.)

Regards,

Nick

rgetz, 2015-08-17:
Ok - but it still an issue for all others - uses should not need to build from source.

I helped Tim get set up on the machine that can re-create the issue that we have - and he will debug things there (I think it is just a library that we don't package that is necessary).

-Robin

pettefar, 2015-08-17:
Do you want any information from me about my build?

Nick

radhermit, 2015-08-17:
So one of the issue appears that macdeployqt isn't handling the plugins correctly, namely the cocoa plugin, when it's creating the dmg. The reason the pre-release works is that I didn't use macdeployqt at all and mostly fixed things via scripting on the command line.

In order to create new dmgs we could just drop new app builds into the original pre-release dmg which appears to work fine until we can figure out what else macdeployqt should be doing.

radhermit, 2015-08-17:
Slightly more information, one of the issues is that the brew install of qt breaks macdeployqt (see https://github.com/Homebrew/homebrew/issues/42191) especially on Qt-5.5.0. Installing directly from upstream qt gets you farther; however, upstream's qml import parsing used by macdeployqt doesn't appear to be working correctly for me or else I'm invoking it wrong since I get lots of "Skip import: name or path is empty" log output (which also appears in the travis-ci log output so it's not working there either) when it should be copying things over. Anyway, manually copying over of the required qml files from the Qt install works and the app and dmg work as expected then.

In summary, either we'll have to script around those issues to automate things (and get brew to fix their install) or just do manual releases for now.

rgetz, 2015-08-17:
Tim:

What is your recommendation?

rgetz, 2015-08-17:
Does https://github.com/aurelien-rainone/macdeployqtfix help?

radhermit, 2015-08-17:
Yep, that scripts works to fix the breakage with using brew's macdeployqt, but it doesn't fix the qml script bundling issues.

radhermit, 2015-08-17:
Really I'm not sure if we want a fully automated release mechanism since there should obviously be a human in the loop at this point if we're not actually also automating any tests to make sure these apps run properly (as seen by the past two releases).

radhermit, 2015-08-18:
On 2015-08-17 17:23, rgetz wrote:
> What is your recommendation?

Should I rebuild a working 0.85 release for OS X then or are things
ready for a 0.86 release?

rgetz, 2015-08-18:
If you have a recipe - I would do the 0.85 version for now.

-Robin

radhermit, 2015-09-11:
Forgot about this for a bit, but a working 0.85 release should now be attached.

In summary, it's now using qt-5.5.0 along with macdeployqt and that python script mentioned above in addition to reverting a previous commit to main.cpp that added the current directory to the library path which messes everything up when qt plugins (such as libcocoa) are searched for.

I'll probably wait for upstream qt/brew to fix their issues with macdeployqt instead of hacking up the travis script to use that python script.

closed 2015-09-11 by radhermit