analogdevicesinc/Pixelpulse2 #145: Windows doesn't work - missing TLS/SSL libraries (maybe)

rgetz, 2015-07-13:
Things don't start properly:

[image]

Any ideas?

lrgetz, 2015-07-13:
Additional error message.

[image]

rgetz, 2015-07-13:
That looks like a graphics issue, so attaching output from dxdiag.

`
------------------
System Information
------------------
Time of this report: 7/13/2015, 13:27:21
       Machine name: LAUREN-PC
   Operating System: Windows 7 Home Premium 64-bit (6.1, Build 7601) Service Pack 1 (7601.win7sp1_gdr.150525-0603)
           Language: English (Regional Setting: English)
System Manufacturer: Dell Inc.
       System Model: XPS L521X
               BIOS: InsydeH2O Version 03.72.24A16
          Processor: Intel(R) Core(TM) i7-3540M CPU @ 3.00GHz (4 CPUs), ~3.0GHz
             Memory: 8192MB RAM
Available OS Memory: 8074MB RAM
          Page File: 6683MB used, 9463MB available
        Windows Dir: C:\Windows
    DirectX Version: DirectX 11
DX Setup Parameters: Not found
   User DPI Setting: Using System DPI
 System DPI Setting: 120 DPI (125 percent)
    DWM DPI Scaling: UnKnown
     DxDiag Version: 6.01.7601.17514 32bit Unicode
  DxDiag Previously: Crashed in DirectShow (stage 1). Re-running DxDiag with "dontskip" command line parameter or choosing not to bypass information gathering when prompted might result in DxDiag successfully obtaining this information

------------
DxDiag Notes
------------
      Display Tab 1: No problems found.
        Sound Tab 1: No problems found.
          Input Tab: No problems found.

--------------------
DirectX Debug Levels
--------------------
Direct3D:    0/4 (retail)
DirectDraw:  0/4 (retail)
DirectInput: 0/5 (retail)
DirectMusic: 0/5 (retail)
DirectPlay:  0/9 (retail)
DirectSound: 0/5 (retail)
DirectShow:  0/6 (retail)

---------------
Display Devices
---------------
          Card name: Intel(R) HD Graphics 4000
       Manufacturer: Intel Corporation
          Chip type: Intel(R) HD Graphics Family
           DAC type: Internal
         Device Key: Enum\PCI\VEN_8086&DEV_0166&SUBSYS_054F1028&REV_09
     Display Memory: 1696 MB
   Dedicated Memory: 64 MB
      Shared Memory: 1632 MB
       Current Mode: 1920 x 1080 (32 bit) (60Hz)
       Monitor Name: Generic PnP Monitor
      Monitor Model: unknown
         Monitor Id: AUO00ED
        Native Mode: 1920 x 1080(p) (60.011Hz)
        Output Type: Internal
        Driver Name: igdumd64.dll,igd10umd64.dll,igd10umd64.dll,igdumd32,igd10umd32,igd10umd32
Driver File Version: 8.15.0010.2778 (English)
     Driver Version: 8.15.10.2778
        DDI Version: 11
       Driver Model: WDDM 1.1
  Driver Attributes: Final Retail
   Driver Date/Size: 9/5/2012 18:59:10, 8089088 bytes
        WHQL Logo'd: Yes
    WHQL Date Stamp: 
[... 245 lines cut ...]
EVR Power Information
---------------
Current Setting: {5C67A112-A4C9-483F-B4A7-1D473BECAFDC} (Quality) 
  Quality Flags: 2576
    Enabled:
    Force throttling
    Allow half deinterlace
    Allow scaling
    Decode Power Usage: 100
  Balanced Flags: 1424
    Enabled:
    Force throttling
    Allow batching
    Force half deinterlace
    Force scaling
    Decode Power Usage: 50
  PowerFlags: 1424
    Enabled:
    Force throttling
    Allow batching
    Force half deinterlace
    Force scaling
    Decode Power Usage: 0

`

dNechita, 2015-07-14:
Might be worth trying to install OpenSSL:
http://slproweb.com/download/Win32OpenSSL_Light-1_0_2d.exe
from page:
http://slproweb.com/products/Win32OpenSSL.html

which should get rid of the SSL related error messages.

-Dan

lrgetz, 2015-07-14:
That did fix it -- things work now.

I assume that we should be providing 64 and 32 bit versions?
http://indy.fulgan.com/SSL/ ?

Thanks
-Robin

dNechita, 2015-07-14:
Since Pixelpulse2 is 32-bit compiled I do not think we need the 64-bit version of the SSL library.

rgetz, 2015-08-07:
Closing this, since it seems to be resolved.

closed 2015-08-07 by rgetz