analogdevicesinc/Pixelpulse2 #152: Cannot start PixelPulse2 0.85 and 0.8 on Windows 10

DanilBorchevkin, 2015-10-14:
Errors:

[image]

For 0.8 version nothing only info about QMutex.

System:
Windows 10 Pro Insider Preview 10.0.10547 10547
Intel(R) Celeron(R) CPU G1820 @ 2.70GHz, 2700 МГц
RAM 12 Gb
Nvidia GeForce GTX 650 Ti

itdaniher, 2015-10-14:
Would you please download distrib.zip and pixelpulse2.exe from https://ci.appveyor.com/project/analogdevicesinc/pixelpulse2/build/0.1.497/artifacts, extract the zip file, place pixelpulse2.exe in the new distrib folder, and run this version?

Your errors look related to the presence of encryption-related files on another drive on your computer. I have not encountered an issue like this before, but recent changes to Pixelpulse2 builds for Windows include the additional cryptography support libraries.

Thanks,
Ian

acozma, 2015-12-10:
Pixelpulse2 v0.87 is now the recommended version to use on Windows 10. Can you please try that to see if it fixes the issues you've been seeing?

Thanks,
Andrei

rgetz, 2016-04-05:
We can't replicate, so closing

closed 2016-04-05 by rgetz