analogdevicesinc/Pixelpulse2 #153: USB drivers not found

vthielen, 2015-11-26:
When I plug in the ADALM1000, after installing the pixelpulse2 software on a windows 7 machine, it doen's find the drivers.
Do they have to be installed separately?

acozma, 2015-12-04:
Normally the drivers are downloaded automatically from the internet when the device is connected to the PC for the first time. Can you please check that you have a working internet connection on that PC and try to update the driver by going to Device Manager -> Other Devices, right click on the ADALM1000 device and select "Update Driver Software".

vthielen, 2015-12-08:
I solved the problem by installing drivers separately. Maybe firewall was
the problem.

Regards,

Bart

closed 2015-12-10 by acozma