analogdevicesinc/Pixelpulse2 #155: Fonts not clear and graph not visible in windows 7

jaganmail, 2015-12-01:
[image]

Fonts not clear and graph update is not visible. The board generates the waveform and its visible by connecting LED. 
Configuration: Windows 7, 4 GB RAM

dNechita, 2015-12-03:
Hi,

It looks like your graphics card may not have all the drivers it needs or it does not support all features that Pixelpulse2 requires.

-Dan

rgetz, 2015-12-03:
Dan:

That's kind of a lame answer. ;)

Can you be more specific? What features are required? Why doesn't Qt fall back to a software render?

jaganmail, 2015-12-03:
Dan: & Rgetz: 
I am just a starter and my observations are
Morning i was trying few other 32 bit windows 7 system and the same problem in all the cases. When, i tried on 64 bit windows 7 it all worked fine. Looks we may have to have a separate win setup for 32 bit windows 7.

Help me

dNechita, 2015-12-03:
>  Looks we may have to have a separate win setup for 32 bit windows 7.
Could be so but first I need to investigate and make sure that is not a problem of graphics support. Could you please provide information about the graphics card on your system that runs window7 32bit?

You can do that by pressing the start button->All Programs->Accessories->System Tools->System Information. Then in the newly opened window click on "Components" then click on "Display". Information about your graphics card should appear in the right side of the window. 

-Dan

jaganmail, 2015-12-04:
[image]

Dan
Details enclosed

dNechita, 2015-12-04:
Pixelpulse2 has just been tested on a Windows 7 32-bit platform and it works properly. Pixelpulse2 is a 32-bit application and it is expected to run on both x86 and x64 architectures. Therefore there is no need for separate installers.
However you should download the installer for Pixelpulse2 again and try it, since we made some modifications to it in order to make sure that only 32 bit dependencies get installed for a x86 architecture and only 64-bit dependencies get installed for a x64 architecture.

Could you please make sure that you have the latest driver installed for the Intel HD Graphics? There is a tool provided by Intel that can help you with that:
http://www.intel.com/p/en_US/support/detect/?iid=subhdr+dnlds_detect

Let me know if you're problem persist.

-Dan

jaganmail, 2015-12-05:
Thank you Dan
Working fine and screen shot enclosed 
[image]

closed 2015-12-05 by jaganmail

dNechita, 2015-12-07:
I'm glad to hear that.

-Dan