analogdevicesinc/Pixelpulse2 #157: Installation problem

AndersLPetersson, 2015-12-04:
Having problem installing Pixelpulse2. Can someone explain, step by step, the installation procedure. Thanks in advance.

rgetz, 2015-12-04:
Anders:

Thanks for reaching out. Before we can help - we need to know more info. What OS? Are you building from source? Or binary?

AndersLPetersson, 2015-12-04:
OK ! I run Win7 64 bit on a Dell Precision T3400. Intel duo. 2.66 GHz. Binary, I believe.

rgetz, 2015-12-04:
Anders:

Thanks for closing - I assume you resolved your issue. Can you tell us what the issue was, and we can try to improve the docs. 

Thanks
- Robin

AndersLPetersson, 2015-12-04:
Sorry. I closed by mistake.  //  Anders

dNechita, 2015-12-07:
Could you please describe the problem you are having?
To install Pixelpulse2 on a windows machine you need to download the pixelpulse2-v0.86_win_setup.exe installer from:
https://github.com/analogdevicesinc/Pixelpulse2/releases
and run it.

-Dan

AndersLPetersson, 2015-12-10:
Dan,
I downloaded Pixelpulse2-v0,86.exe,Pixelpulse2-v0,86_win_setup.exe and window-depens-v0.86.zip to a folder. Unzipped the windows-depens file. 
Ran the setup. Plugged in the Adalm1000 and started Pixelpulse. An empty frame opens. The installation looks like this:

//  Anders

From: Dan Nechita 
Sent: Monday, December 07, 2015 9:57 AM
To: analogdevicesinc/Pixelpulse2 
Cc: AndersLPetersson 
Subject: Re: [Pixelpulse2] Installation problem (#157)

Could you please describe the problem you are having?
To install Pixelpulse2 on a windows machine you need to download the pixelpulse2-v0.86_win_setup.exe installer from:
https://github.com/analogdevicesinc/Pixelpulse2/releases
and run it.

-Dan

dNechita, 2015-12-10:
Hi Anders,

Unfortunately the image you have attached to your reply message did not make its way to your comment on git issues: https://github.com/analogdevicesinc/Pixelpulse2/issues/157
To be sure the images gets attached you can comment directly on the issues section of Pixelpulse2 git repository using an internet browser.

You only need the pixelpulse2-v0.86_win_setup.exe installer. You don't need the other files you've mentioned in your previous reply.

Since I don't know exactly how the empty frame looks like I will assume the only problem is that the drivers for your ADALM1000 have not installed properly.

To check if  the driver for ADALM1000 has installed properly, open Device Manager and check for 'Universal Serial Bus devices', expand it and you should see a 'WinUsb Device' item. If you do then the driver is installed properly.

If you don't, it means that the driver is not installed and instead you should look for Device Manager -> Other Devices->ADALM1000. If this is the case, right click on the ADALM1000 device and select "Update Driver Software".

-Dan

AndersLPetersson, 2015-12-10:
Hi Dan

I reeinstalled using only the setup file. Same result.

I looked for the driver. I have a “WinUsb Generic Device” item.

The driver was installed using the Zadig program. It detected the Adalm1000 and asked if a driver
should be installed. I clicked OK. Maybe the wrong driver was installed.

//  Anders

From: Dan Nechita 
Sent: Thursday, December 10, 2015 10:27 AM
To: analogdevicesinc/Pixelpulse2 
Cc: AndersLPetersson 
Subject: Re: [Pixelpulse2] Installation problem (#157)

Hi Anders,

Unfortunately the image you have attached to your reply message did not make its way to your comment on git issues: #157
To be sure the images gets attached you can comment directly on the issues section of Pixelpulse2 git repository using an internet browser.

You only need the pixelpulse2-v0.86_win_setup.exe installer. You don't need the other files you've mentioned in your previous reply.

Since I don't know exactly how the empty frame looks like I will assume the only problem is that the drivers for your ADALM1000 have not installed properly.

To check if the driver for ADALM1000 has installed properly, open Device Manager and check for 'Universal Serial Bus devices', expand it and you should see a 'WinUsb Device' item. If you do then the driver is installed properly.

If you don't, it means that the driver is not installed and instead you should look for Device Manager -> Other Devices->ADALM1000. If this is the case, right click on the ADALM1000 device and select "Update Driver Software".

-Dan

dNechita, 2015-12-10:
Can you share a screenshot?

-Dan

AndersLPetersson, 2015-12-10:
Here are a few screenshots showing the empty frame, the USB and the errors shown when running the
previous version (0.85). // Anders

From: Dan Nechita 
Sent: Thursday, December 10, 2015 12:08 PM
To: analogdevicesinc/Pixelpulse2 
Cc: AndersLPetersson 
Subject: Re: [Pixelpulse2] Installation problem (#157)

Can you share a screenshot?

-Dan

dNechita, 2015-12-10:
I'm not receiving any of your images. Use below link to take a look at your replies.
<pre>
https://github.com/analogdevicesinc/Pixelpulse2/issues/157
</pre>
Please use the browser to post a reply that contains images.

-Dan

AndersLPetersson, 2015-12-10:
Ok Dan, here are the files.
[image]
[image]
[image]

dNechita, 2015-12-10:
Thanks! The images are really helpful.
Your usb driver is properly installed so you don't have to worry about it.

The issue your having usually rises when the graphics card is old enough so that it doesn't support OpenGL 3.0 which is required to for the Pixelpulse 2 to work or the graphics card is using only a basic set of drivers (which usually provides OpenGL 1.1) and it needs the latest drivers. Since I don't know what graphics card your system has I don't know which of the two cases described above apply to you.

You can get information about the graphics card by pressing the start button->All Programs->Accessories->System Tools->System Information or right-click on desktop->Screen Resolution->Advanced Settings and click on the Adapter tab.

-Dan

AndersLPetersson, 2015-12-11:
Hi Dan

My graphics card is a NVIDIA Quadro NVS 290 supporting OpenGL 2.1.  I will get a new card whithin a few days
and then come back with the result.

Thanks!  //  Anders

From: Dan Nechita 
Sent: Thursday, December 10, 2015 4:57 PM
To: analogdevicesinc/Pixelpulse2 
Cc: AndersLPetersson 
Subject: Re: [Pixelpulse2] Installation problem (#157)

Thanks! The images are really helpful.
Your usb driver is properly installed so you don't have to worry about it.

The issue your having usually rises when the graphics card is old enough so that it doesn't support OpenGL 3.0 which is required to for the Pixelpulse 2 to work or the graphics card is using only a basic set of drivers (which usually provides OpenGL 1.1) and it needs the latest drivers. Since I don't know what graphics card your system has I don't know which of the two cases described above apply to you.

You can get information about the graphics card by pressing the start button->All Programs->Accessories->System Tools->System Information or right-click on desktop->Screen Resolution->Advanced Settings and click on the Adapter tab.

-Dan

dNechita, 2015-12-11:
Hi Anders,

Found some web pages that confirm that your graphics card supports OpenGL 2.1 but I've also found other web pages/articles that suggest that NVIDIA Quadro NVS 290 supports OpenGL 3.3.
https://www.techpowerup.com/gpudb/1453/quadro-nvs-290.html
http://www.gpuzoo.com/GPU-NVIDIA/Quadro_NVS_290.html
http://www.game-debate.com/hardware/index.php?gid=1754&graphics=Quadro%20NVS%20290

You could give it a try and download latest drivers from NVIDIA:
http://www.nvidia.com/download/index.aspx
After that you could run OpenGL Extensions Viewer:
http://www.realtech-vr.com/glview/
to check if OpenGL 3.3 is indeed supported by your graphics card.

-Dan

AndersLPetersson, 2015-12-12:
Dan,
After some problems with the updating of Adalm1000 everything now works fine. It was possible to update the NVS 290 to OpenGL 3.3 (the card is 7 years old !!). Thanks for your kind support !!
//  Anders

dNechita, 2015-12-14:
Glad to hear that everything worked out.

-Dan

closed 2015-12-14 by dNechita