analogdevicesinc/Pixelpulse2 #158: Qt: Session management error on Debian GNU/Linux 8.2 (jessie)

milnepe, 2015-12-28:
I've followed the Debian build instructions which completed ok. I'm getting this runtime error:
root@inspiron:/home/pmilne/tmp/pixelpulse2# ./pixelpulse2 
Qt: Session management error: None of the authentication protocols specified are supported
Using libusb hotplug
device insert 0x1649110
QQmlApplicationEngine failed to load component
qrc:/qml/main.qml:133 Type DeviceManagerPane unavailable
qrc:/qml/DeviceManagerPane.qml:5 module "QtQuick.Controls.Styles" version 1.3 is not installed

Here are the packages I have installed:
root@inspiroot@inspiron:/home/pmilne/tmp/pixelpulse2# dpkg -l | grep qml
ii  libqt5qml5:amd64                        5.3.2-4	amd64	Qt 5 QML module
ii  qml                                     5.3.2-4	amd64	Qt 5 QML viewer
ii  qml-module-qtqml-models2:amd64          5.3.2-4	amd64	Qt 5 Models2 QML module
ii  qml-module-qtquick-controls:amd64       5.3.2-2	amd64	Qt 5 Quick Controls QML module
ii  qml-module-qtquick-dialogs:amd64        5.3.2-2	amd64	Qt 5 Dialogs QML module
ii  qml-module-qtquick-layouts:amd64        5.3.2-2	amd64	Qt 5 Quick Layouts QML module
ii  qml-module-qtquick-privatewidgets:amd64 5.3.2-2	amd64	Qt 5 Private Widgets QML module
ii  qml-module-qtquick-window2:amd64        5.3.2-4	amd64	Qt 5 window 2 QML module
ii  qml-module-qtquick2:amd64               5.3.2-4	amd64	Qt 5 Qt Quick 2 QML module
ii  qmlscene                                5.3.2-4	amd64	Qt 5 QML scene viewer
ii  qtquick1-qml-plugins:amd64              5.3.2-3	amd64	Qt 5 QtQuick1 qml plugins
ii  qtquick1-qmltooling-plugins:amd64       5.3.2-3	amd64	Qt 5 QtQuick1 qmltooling plugins

dNechita, 2016-01-06:
Hi,

Regarding:
>QQmlApplicationEngine failed to load component
qrc:/qml/main.qml:133 Type DeviceManagerPane unavailable
qrc:/qml/DeviceManagerPane.qml:5 module "QtQuick.Controls.Styles" version 1.3 is not installed

You need Qt 5.4 or higher to build Pixelpulse2.
To install it, please take a look at step: <b>Download and install Qt5.4</b> from:
https://github.com/analogdevicesinc/Pixelpulse2/blob/master/README.md

-Dan

rgetz, 2016-04-01:
Closing this

closed 2016-04-01 by rgetz