analogdevicesinc/Pixelpulse2 #159: Crash on close

cyrozap, 2016-02-07:
I get this error when closing the application if no devices were ever plugged in, followed by a backtrace:

```
Error in `pixelpulse2': corrupted double-linked list
```

Since it happens right after it prints `Closing devices`, my first guess is that it's trying to free a non-existent device, but I haven't actually looked at the code yet.

rgetz, 2016-02-07:
Thanks for the bug report. 

Can you detail pixelpulse version, and operating system, so we can replicate?

Thanks
- Robin

rgetz, 2016-03-23:
I can't replicate on the latest version (on Windows Built: 2016-03-10    Version: v0.88-1-g2a3ff9b). Can someone else try Linux & OS-x (and close if this can't be replicated?)

Thanks
-Robin

cyrozap, 2016-03-24:
Sorry I never replied to your previous post! I'm running v0.88-8-g2540d52 on Arch Linux (built from the [AUR package](https://aur.archlinux.org/packages/pixelpulse2-git)) and I still get this issue, but if I leave the application open for a while it sometimes exits gracefully.

rgetz, 2016-03-24:
Do you get a full backtrace?

radhermit, 2016-04-01:
So in a different manner, SIGABRT is thrown if pixelpulse appears to be closed after a successful release query response is received. If it's closed after the request is sent, but before a response is received then nothing crashes, see the attached backtrace.

I've also seen the reported corrupted double-linked list issue, but I can't reliably reproduce that in the same way at the moment.

[backtrace-release-request-sigabrt.txt](https://github.com/analogdevicesinc/Pixelpulse2/files/200508/backtrace-release-request-sigabrt.txt)

cyrozap, 2016-04-01:
Here's my backtrace as well:

```
Thread 1 "pixelpulse2" received signal SIGABRT, Aborted.
0x00007ffff4d6d2a8 in raise () from /usr/lib/libc.so.6
(gdb) bt
#0  0x00007ffff4d6d2a8 in raise () from /usr/lib/libc.so.6
#1  0x00007ffff4d6e72a in abort () from /usr/lib/libc.so.6
#2  0x00007ffff4da9369 in __libc_message () from /usr/lib/libc.so.6
#3  0x00007ffff4daed96 in malloc_printerr () from /usr/lib/libc.so.6
#4  0x00007ffff4daf061 in malloc_consolidate () from /usr/lib/libc.so.6
#5  0x00007ffff4dafc00 in _int_free () from /usr/lib/libc.so.6
#6  0x00007ffff48abe3b in hb_blob_destroy () from /usr/lib/libharfbuzz.so.0
#7  0x00007ffff48bc270 in ?? () from /usr/lib/libharfbuzz.so.0
#8  0x00007ffff48b16e6 in hb_face_destroy () from /usr/lib/libharfbuzz.so.0
#9  0x00007ffff48b4cab in hb_font_destroy () from /usr/lib/libharfbuzz.so.0
#10 0x00007ffff700a340 in QFontEngine::~QFontEngine() () from /usr/lib/libQt5Gui.so.5
#11 0x00007fffec94c199 in QFontEngineFT::~QFontEngineFT() () from /usr/lib/libQt5XcbQpa.so.5
#12 0x00007ffff700a5ed in QFontEngineMulti::~QFontEngineMulti() () from /usr/lib/libQt5Gui.so.5
#13 0x00007fffec90cab9 in ?? () from /usr/lib/libQt5XcbQpa.so.5
#14 0x00007ffff6ffa37d in QFontPrivate::~QFontPrivate() () from /usr/lib/libQt5Gui.so.5
#15 0x00007ffff6ffb5ef in QFont::~QFont() () from /usr/lib/libQt5Gui.so.5
#16 0x00007fffec914021 in ?? () from /usr/lib/libQt5XcbQpa.so.5
#17 0x00007fffec913fe3 in ?? () from /usr/lib/libQt5XcbQpa.so.5
#18 0x00007ffff6f495b1 in QGuiApplicationPrivate::~QGuiApplicationPrivate() () from /usr/lib/libQt5Gui.so.5
#19 0x00007ffff6f49769 in QGuiApplicationPrivate::~QGuiApplicationPrivate() () from /usr/lib/libQt5Gui.so.5
#20 0x00007ffff60e056c in QObject::~QObject() () from /usr/lib/libQt5Core.so.5
#21 0x00007ffff60adf82 in QCoreApplication::~QCoreApplication() () from /usr/lib/libQt5Core.so.5
#22 0x0000000000408e88 in main ()
```

jonpearson, 2016-04-06:
Not an issue for the labs, but should be resolved relatively quickly.