analogdevicesinc/Pixelpulse2 #161: firmware update issues on Linux

radhermit, 2016-02-29:
Currently on Linux the updating method tries to download the firmware file to /firmware.bin in the root dir specifically. This is due to getTmpPathForFirmware() returning an empty string for Linux.

It seems there are a few better ways to handle this in the same fashion across all supported OS's. One option for the installed case would be to create an application dir and download the firmware to that file using versioned files so the user could upgrade/downgrade as they wish. Another option (perhaps a fallback for the non-installed case?) would be to use a more cross-platform friendly method such as using QDir::tempPath() or similar to get a path to the OS specific tempdir and then create a templated tmpdir in there such as XXXXXX-m1k-firmware and then toss the downloads into that.

jonpearson, 2016-04-06:
Anything that makes the device not able to be updated is a problem for the labs, so this is a high priority in this regard.  The worst case is when the M1K becomes non-functional.  It would be good to make the firmware update process as bullet-proof as possible.

dNechita, 2016-04-07:
With the following commit: https://github.com/analogdevicesinc/Pixelpulse2/commit/080e4412140a1e55932f65e6627b95063be3e2ab
Pixelpulse2 will store the firmware in the temporary directory of a file system (which can vary on different operating systems). There it will create pixelpulse2 dir and inside it the m1k_firmware dir where the firmware file will be downloaded.

luciansin, 2016-05-13:
bug fixed and closed.

closed 2016-05-13 by luciansin