analogdevicesinc/Pixelpulse2 #162: failed firmware update on Linux appears to make the device not able to be updated on Linux

radhermit, 2016-02-29:
If the firmware update fails on Linux for whatever reason, e.g. permissions issues when trying to download the file to root as seen in #161, then the device appears to enter a state where it can't be updated by pixelpulse or even manually via bossa/bossac on Linux.

The reason for this is that the USB interface keeps going up and down as seen below from some dmesg output.

```
[3037180.498164] cdc_acm 1-1.6:1.0: ttyACM1: USB ACM device
[3037180.727650] usb 1-1.6: USB disconnect, device number 96
[3037180.904340] usb 1-1.6: new high-speed USB device number 97 using ehci-pci
[3037180.990067] usb 1-1.6: New USB device found, idVendor=03eb, idProduct=6124
[3037180.990072] usb 1-1.6: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[3037180.990596] cdc_acm 1-1.6:1.0: ttyACM1: USB ACM device
[3037181.239702] usb 1-1.6: USB disconnect, device number 97
[3037181.417413] usb 1-1.6: new high-speed USB device number 98 using ehci-pci
[3037181.693258] usb 1-1.6: new high-speed USB device number 99 using ehci-pci
[3037182.174266] usb 1-1.6: device not accepting address 99, error -71
[3037182.373238] usb 1-1.6: new high-speed USB device number 101 using ehci-pci
```

It does appear possible to reinstall a working firmware from Windows when this occurs using bossa.

itdaniher, 2016-02-29:
Could be power-related, see https://github.com/analogdevicesinc/m1k-cad/issues/4.

radhermit, 2016-02-29:
Yep, I guess this is basically the same thing as https://github.com/analogdevicesinc/m1k-fw/issues/3 and using that python script works to update things when straight bossac (which is what pixelpulse uses) fails.

It seems like pixelpulse should implement the same thing so people don't have to use the python script.

dNechita, 2016-04-05:
I am fixing this by making sure not to put the device in programming mode until I'm sure that the firmware has successfully downloaded and the file is accessible.
Commit: https://github.com/analogdevicesinc/Pixelpulse2/commit/69470cd309794fc172d00bfd9f599b17ecc04983

jonpearson, 2016-04-06:
Same as for #161.

msuciu, 2016-04-27:
This bug is now fixed. The firmware.bin is no longer downloaded in root so the permission issues are now solved, as Pixelpulse stores firmware in a temporary directory. 

Running find -name "firmware.bin" from root, outputs the following:

./tmp/pixelpulse2/m1k_firmware/firmware.bin

closed 2016-04-27 by msuciu