analogdevicesinc/Pixelpulse2 #163: X/Y axis labels wrong on os-x

rgetz, 2016-03-23:
in the X/Y plots - the labels were being shown as 0.00000000001 - which consumed all the screen space. Only happens on OS-X.

-Robin

While updating that code - Also noticed that axises don't use the same number of digits (which they should).

For example:

[image]

should be 0.20, 0.15, 0.10, 0.05, 0.00, etc

luciansin, 2016-04-12:
Same number of digits fixed in Windows.
It was tested with v0.89 and on Win7.32bits, Win7.64bits, Win10.32bits and Win10.64bits.

rgetz, 2016-04-12:
The original issue reported was X/y plots on OS-X. 

Does that still occur?

luciansin, 2016-04-12:
X/Y plots on OS-X work now. Bug fixed and closed.

closed 2016-04-12 by luciansin

rgetz, 2016-04-12:
It wasn't that they didn't work, it was that the printed out too many decimal places. 0.1 was "0.1000000001"

luciansin, 2016-04-12:
Now the maximum number of decimals is 4.