analogdevicesinc/Pixelpulse2 #165: Sometimes (?) Pixelpulse doesn't draw on the complete screen

rgetz, 2016-04-01:
[image]

happens on latest version.

I'm not sure if this is because something is getting confused about a device manager, or something else being on the screen, when it is really not?

Thanks

rgetz, 2016-04-01:
To duplicate : When not using "repeated sweep", and using the "10s - sample time" option, the oscilloscope windows is never filled.

luciansin, 2016-04-05:
I couldn't reproduce this bug on Windows 7(64bits) and PixelPulse (Version: v0.88-1-g2a3ff9b). But I've found another bug starting from the above settings. First not use "repeated sweep" and use "10s - sample time option". Afterwards if you change to "1s - sample time option", the application will be blocked.

rgetz, 2016-04-05:
I couldn't make this happen 100% of the time. But can make it happen eventually. 

It is like one the device manage thinks it should be drawn, but isn't.

luciansin, 2016-04-05:
You are right. It happened also to me eventually.

jonpearson, 2016-04-06:
Could not duplicate, but probably due to not enough tries as others have noted.  This is not a high priority for the summer camp labs, but should be resolved quickly nonetheless.

dNechita, 2016-04-08:
Fixed in commit: https://github.com/analogdevicesinc/Pixelpulse2/commit/c0d0ca2a4783659c0190510e205a7322d43ee205

I was able to always reproduce the bug if I did the following:
1.start app
2.uncheck Repeated Sweep option
3.set any channel to output a periodic voltage signal - just for a better visualization  of the issue
4. start capturing
5. set the Sample Time to a higher value than the one currently set.

luciansin, 2016-04-12:
90% fixed 
There is a small empty area at the beginning of the plot where the signal is not drawn at all 
when choosing an upper sample time. After all the plot is filled in, this empty area disappears.

AlexandraTrifan, 2019-07-02:
Could not reproduce the issue. 
Will close the bug.

closed 2019-07-02 by AlexandraTrifan