analogdevicesinc/Pixelpulse2 #168: Backspace key doesn't work

rgetz, 2016-04-01:
It is not possible to clear the first characters of values using the Backspace key. 
It is possible if I use delete instead of backspace.

[image]

luciansin, 2016-04-05:
I couldn't reproduce this on Windows 7(64bits) and PixelPulse v0.88-1-g2a3ff9b. What OS and PixelPulse version did you use?

jonpearson, 2016-04-06:
Could not reproduce.  Other things noticed:  All digits are not displayed if the value is not changed, i.e. if the voltage reads 2.5000, then all digits are erased by backspacing, then 2.5 is entered, the trailing zeros do not show up; also, the frequency reading goes into scientific notation -- while not a problem, it is a little inconvenient.

AlexandraTrifan, 2016-04-08:
Regarding the first part of the issue: could not reproduce. This issue was solved in the following commit: https://github.com/analogdevicesinc/Pixelpulse2/commit/40cf4aa07355f802896fbe4617093e3a85754738 .
As for the second part, the frequency is now displayed without the scientific notation.

luciansin, 2016-04-12:
part1 - could not reproduce
part2 - fixed
This bug will be closed.  It was tested with v0.89 and on Win7.32bits, Win7.64bits, Win10.32bits and Win10.64bits.

closed 2016-04-12 by luciansin