analogdevicesinc/Pixelpulse2 #172: Pixelpulse2 Waveform Glitches

jonpearson, 2016-04-06:
I've seen glitches on the generated waveforms while making videos, and originally attributed them to an interaction between Pixelpulse and Camtasia (and possibly partially due to problems with my old PC), but have recently observed them on a new PC when not running Camtasia.  Some examples are shown in the following screenshots.  I had the system set up to source voltage on Channel A and measure voltage on Channel B.

[image]
[image]
[image]
[image]

dNechita, 2016-04-26:
I have seen this behavior on systems where there was no graphic acceleration (such as virtual machines) but Qt managed to switch to a software graphic renderer.
It would help to know what kind of graphic renderer this system has and which version of OpenGL the renderer supports.
@jonpearson could you please use [glview](http://www.realtech-vr.com/glview/) to get this kind of information and post it here?
You should look for the first opengl version that doesn't show 100% feature complete, like version 4.2 in the picture below and expand it when taking the screenshot. Also the information from the panel on the right is useful.
[image]

Thank you,
-Dan

jonpearson, 2016-04-26:
Hi Dan,

Here's the screenshot:

[image]

Thanks.

--Jonathan

dNechita, 2016-04-27:
Thanks Jonathan!

It looks that your system has more than enough graphic resources. I will keep investigating.

Once you start seeing these glitches, do they persist for a while? or the glitch happens once in a while for a very short period of time?

jonpearson, 2016-04-27:
Hi Dan,

They occur for short time intervals, but keep happening over and over.  It took a number of tries to get a reasonable screen capture because their occurrence was random and short-lived.  Their frequency was variable as well.  When using Camtasia they got much more frequent when I began recording the screen.

Hope this helps.

--Jonathan

luciansin, 2016-05-05:
Hi all,
If you want to more frequent a glitch you can make a full screen window, activate X-Y plots, run play and then activate any menu from the window (I kept activated the settings menu). See attached picture:
[image]