analogdevicesinc/Pixelpulse2 #173: Pixelpulse2 Trace Latchup When M1K Unplugged/Replugged

jonpearson, 2016-04-06:
With the system running normally, if the M1K USB cable is unplugged, then plugged back in, the traces freeze when the arrow button is pressed.  The only way I could find to get out of this condition was to close and reopen Pixelpulse2.  The trace latchup condition is illustrated in the following screenshot.

[image]

rgetz, 2016-04-08:
I think the easiest way to replicate this would be to get something like:

https://www.adafruit.com/products/1620?gclid=Cj0KEQjwrZ24BRC098fr-OqnuMkBEiQAKQ9lgERSbIVjgZgWq1DD1uHZjbS7soIeux4Ny3F0DdNHyvMaAtCs8P8HAQ

and sit there clicking switches. If you need one - please let me know (I ordered some).
-Robin

cristi-iacob, 2019-08-20:
We have solved this bug and the fix will be available in the next version of the application.

closed 2019-08-20 by cristi-iacob