analogdevicesinc/Pixelpulse2 #174: sometimes with repeated sweep unchecked, pixelpulse gets blocked

luciansin, 2016-04-12:
Sometimes when repeated sweep is unchecked, the capture is on with any periodic signal and I change sample time, the application gets frozen with this message appeared in a terminal:
"addr2line is not recognized". 
I wasn't able to read all the message because the terminal became white and pixelpulse was not responding anymore.

jonpearson, 2016-04-14:
This sounds similar to what I found when checking the long sample time/not filling the screen bug.  I got the same error ("'addr2line' is not recognized as an internal or external command, operable program or batch file.") changing the sample time from 10 s to 1 s.  It happened almost every time when I changed the time scale before changing the sample time to 1 s.  The sequence was:

Launch PP
Press play
Source V
Choose waveform
Change sample time to 10 sec
Wait for screen to fill
Change the displayed time scale
Change sample time to 1 sec
Crash

This happened in both repeated and non-repeated sweep modes.

luciansin, 2016-04-26:
Using the steps described by Jonathan I was also able to replicate the issue in both repeated and non-repeated modes.

AlexandraTrifan, 2016-05-05:
Fixed with commit 958465734cbcc53d7ba903ba64072f9803b5ee52.

luciansin, 2016-05-12:
Bug found, retested and fixed on Win7.64bits. Bug closed.

closed 2016-05-12 by luciansin