analogdevicesinc/Pixelpulse2 #178: able to set infinite hertz for waveforms

radhermit, 2016-04-26:
As seen in the image below, the draggable points disappear when infinite hertz is set (divide by 0). After poking around a bit, I'm not sure if it's possible to get them to appear again without restarting the program.

This probably should be restricted with limits so the waveform interval can never be set to exactly zero.

[image]

luciansin, 2016-05-12:
I took a look at this bug and it doesn't seem to be fixed. I was still able to reach Infinity(a horizontal line and dots disappeared). The only difference was that after I reached Infinity and I let go of right click of the mouse, the frequency displayed was 20KHz. But the horizontal line was still there without any dot, so I wasn't able to see or to return to a periodic signal anymore.

AlexandraTrifan, 2016-05-13:
The displayed frequency value can't be greater than 20KHz (no matter if the dot is dragged or already released).
Fixed with commit 4a9221c518778e1922ab47b092bc8894b00d5b23.

luciansin, 2016-05-31:
Bug found, retested and fixed on Win7.64bits. Bug closed.

closed 2016-05-31 by luciansin