analogdevicesinc/Pixelpulse2 #181: exported data from a stairstep doesn't reach its minimum value in excel file

luciansin, 2016-05-04:
When you export data from a stairstep waveform, the signal exported to excel doesn't reach its minimum value.

dNechita, 2016-05-05:
Which were the boundaries of the amplitude signal that you set and what were the min and max values that you seen written to the .csv file? This information is required in order to decide if this is a bug or not.

I tried to replicate this issue by setting a stairstep signal with a range from 0.0 V to 5.0 V, 25.0 Hz frequency.
By setting the upper limit to 5.0 V, the received value was of approximately: 4.984 V resulting in a percentage error of approximately: 0.32% (for the device I used). I made an export to .csv and found that the values match with the ones drawn on the plot.
[image]
[image]

On the lower limit there is indeed a problem since the last stair step never reaches the lower limit which in this case is 0 V. I believe we should take a look at how the transmit samples are being generated (get_sample() function from libsmu.hpp). This issue might affect all types of signals but it just happened to be visible only for the stairstep signal.

To sum up:
- there is no problem with the maximum value
- minimum is not being reached for the stairstep signal
- doesn't look to be any problems with the exporting to .csv functionality

luciansin, 2016-05-06:
The boundaries used by me are from 0.5V to 3.5V and the minimum was 0.8V so one step was missing. But if you set the signal from 3.5V to 0.5V then the maximum value is 3.2V. You are right that there are no problems with exporting to csv functionality, but only when I exported to csv I noticed the bug, and this is still present also in the plot window. In the attached txt files, 0A_Voltage doesn't reach its minimum 0.5V value and 0B_Voltage doesn't reach its maximum 3.5V value.

[0A_Voltage.txt](https://github.com/analogdevicesinc/Pixelpulse2/files/251642/0A_Voltage.txt)
[0B_Voltage.txt](https://github.com/analogdevicesinc/Pixelpulse2/files/251643/0B_Voltage.txt)

[image]

dNechita, 2016-05-10:
Fixed with commit: https://github.com/analogdevicesinc/libsmu/commit/206a66466867792b3b45ab2ff9c2a439224d42e5

luciansin, 2016-05-12:
Bug found, retested and fixed on Win7.64bits. Bug closed.

closed 2016-05-12 by luciansin