analogdevicesinc/Pixelpulse2 #185: Voltage Measurement Anomaly

jonpearson, 2016-05-27:
When sourcing a voltage on Channel A, passing it through a network, and measuring the output voltage on Channel B I observed the following anomaly.  It can be seen on measured current as well.
[image]
Changing the time scale works as a workaround, but it would be good to get this resolved soon for the labs.

dNechita, 2016-07-14:
To solve this a new feature was added. The feature allows user to set an amount of time which will delay the acquisition.

This was added in commit: https://github.com/analogdevicesinc/Pixelpulse2/commit/67a1cb481ec9731390fedfd3c0a038eda4847686

luciansin, 2016-07-18:
enhancement complete and closed.

closed 2016-07-18 by luciansin