analogdevicesinc/Pixelpulse2 #187: Waveform Measurement Enhancement

jonpearson, 2016-05-27:
It would be useful in the labs to have a way to read values directly from the displayed waveforms rather than estimate them from gridlines, and to be able to display common waveform parameters such as peak-to-peak value.  As an example, in the filter lab it is necessary to find the 3 dB frequency, and it would be good to be able to display the output peak-to-peak voltage to determine this.

dNechita, 2016-05-31:
Peak-to-peak, frequency, RMS (maybe).
Even if peak-to-peak and frequency are being shown when sourcing current or voltage it is required to have the measurements for the setup when voltage is being measured (high impedance).

-Dan

AlexandraTrifan, 2016-06-14:
Enhancement is now complete. The corresponding commit is  28adda26c2d59e008e1e836fb5bff5de7a282cd9.

luciansin, 2016-07-18:
Peak-to peak and RMS values added. Do we want something else added as well?

luciansin, 2016-07-19:
enhancement complete and closed.

closed 2016-07-19 by luciansin