analogdevicesinc/Pixelpulse2 #189: bad plot when restoring a session in maximum size window

luciansin, 2016-06-01:
steps to reproduce:
1. save session from a default size Pixelpulse window like this: 
[image]
2. close and restart application and maximaze Pixelpulse window
3  restore previous saved session. It will look like this, the signal won't fill the entire window:
[image]

AlexandraTrifan, 2016-06-03:
Fixed with commit 1c468d3a40d050e41355361edda36c800a387bd2.

luciansin, 2016-06-14:
Bug retested and fixed in ubuntu 16.04 LTS. 
Not able to retest in windows and mac os yet.

luciansin, 2016-07-18:
Bug retested and fixed also on windows and mac os. Bug closed.

closed 2016-07-18 by luciansin