analogdevicesinc/Pixelpulse2 #190: restore session doesn't restore X-Y plots

luciansin, 2016-06-01:
steps to reproduce:
1. save a session with X-Y plots checked
2. restore that session

in restored session X-Y plots won't be checked. (it's the same with about and device manager)

AlexandraTrifan, 2016-06-03:
This is not necessarily a bug because pixelpulse did not have this feature. It can be labeled as an enhancement. 
Do we want this to be implemented?

AlexandraTrifan, 2016-06-07:
Enhancement complete. (commit c2f8ade751c4cd9323987bbf6816209d572cc42c)
Only the X-Y plots option is restored now. Do we want to restore about and device manager as well?

luciansin, 2016-07-18:
X-Y plots option is restored now. In order be able to close this enhancement and consider it complete, someone please answer Alexandra question.

rgetz, 2016-07-18:
No - device manager, and about can stay closed.

luciansin, 2016-07-18:
enhancement complete and closed.

closed 2016-07-18 by luciansin