analogdevicesinc/Pixelpulse2 #193: pixelpulse stops responding sometimes

luciansin, 2016-06-03:
Sometimes when not using "repeated sweep", and changing from "1s - sample time" to "10s - sample time", pixelpulse stops responding.

luciansin, 2016-06-10:
In order to better reproduce the bug, you can zoom in then zoom out maximum in time domain.

luciansin, 2016-07-22:
This happens also with any periodic signal when with repeated sweep checked, you change sample time to 10s, wait for the screen to fill in then uncheck repeated sweep. After these actions Pixelpulse will stop responding. I think the application "doesn't like" the combination repeated sweep unchecked and sample time 10s.

cristi-iacob, 2019-08-19:
Couldn't reproduce the bug on Windows 10 and Ubuntu 18.04.

closed 2019-08-19 by cristi-iacob