analogdevicesinc/Pixelpulse2 #196: sometimes pixelpulse firmware fails to update

luciansin, 2016-06-09:
steps to reproduce:
1.	Plug M1K in PC.
2.	Close pixelpulse
3.	Open bossa and write an older firmware version.
4.	Close bossa.
5.	Unplug and replug the device in PC. Pixelpulse will automatically open.
6.	Press on “update firmware” button: update will fail.
7.	Close and reopen Pixelpulse(keep the device pluged), then press on “update firmware” button: update will work.

for windows,64bits:
[image]

for windows,32bits:
[image]

dNechita, 2016-06-28:
Fixed in https://github.com/analogdevicesinc/Pixelpulse2/commit/c9f0a5873d0b862bf8de6fc4f4f1c6684ba7f751

luciansin, 2016-07-18:
bug fixed and closed.

closed 2016-07-18 by luciansin