analogdevicesinc/Pixelpulse2 #198: white active menu in ubuntu 16.04 LTS

luciansin, 2016-06-13:
steps to reproduce:
choose any option from any menu from pixelpulse window then activate that menu again

actual behavior:
Active menu is white. After a new right click the menu will be displayed correctly.

This happens for all type of menus in pixelpulse window:
1.settings menu:
[image]
2.channel type menu:
[image]
3.waveform type menu:
[image]

This only happens in Linux (Ubuntu 16.04 LTS)

luciansin, 2016-06-13:
the warning in terminal is: 
libpng warning: iCCP: known incorrect sRGB profile

luciansin, 2016-06-14:
same thing happens with Sample Time menu.

cristi-iacob, 2019-08-19:
Couldn't reproduce the bug on Windows 10 and Ubuntu 18.04.

closed 2019-08-19 by cristi-iacob